ArduPilot/ardupilot #7137: VTOL plane should not report itself as MAV_TYPE_FIXED_WING

janis-github, 2017-10-27:
#### Issue details
_VTOL plane reports itself as MAV_TYPE_FIXED_WING _

When VTOL functionality is enabled via Q_ENABLE=1 the vehicle should report itself as VTOL capable in heartbeat as well.
Currently it reports itself as MAV_TYPE_FIXED_WING.
It should send one of vtol types instead:
MAV_TYPE_VTOL_DUOROTOR
MAV_TYPE_VTOL_QUADROTOR
MAV_TYPE_VTOL_TILTROTOR

#### Version
_3.8.1_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_Quadplane_

#### Hardware type
_SITL_

magicrub, 2017-10-27:
>It should send one of..

Nice observation! This is already implemented with the Q_MAV_TYPE param but I doubt anyone changes it and the default behavior could probably use some changes. The default value of 0 is "AUTO" but always sends MAV_TYPE_FIXED_WING, Herein lies the problem that you stumbled cross in your description. It should send one.. but which one? We are forced to rely on the pilot/user to dig through all the params and change it which is easy to miss.

I suppose we could dig through params and try and figure out which one to send... Do you have any suggestions?

Here's the offending code area:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/quadplane.cpp#L2427

janis-github, 2017-10-28:
Thanks for telling about Q_MAV_TYPE. It fixes my issue but as you say it is not very obvoius.
After browsing through code I'd suggest to derive it from Q_FRAME_CLASS when Q_MAV_TYPE is set to 0. Because that parameter is essential for the VTOL to work correctly.
Like the function Copter::get_frame_mav_type() does.

OXINARF, 2017-11-01:
Q_MAV_TYPE was added only 10 days ago, I'm sure @tridge thought about its implications. I can guess that the main reason for not doing automatic detection is that the available MAVLink VTOL types doesn't cover all the ArduPilot supported VTOL types.

Closing this as it's already done.

closed 2017-11-01 by OXINARF