ArduPilot/ardupilot #7138: Polygon Fence fails without warning after invalid fence upload

Shoob189, 2017-10-27:
#### Issue details
Polygon Geofence is non-functional without any warning if an invalid polygon/return point is loaded. 

As far as I can tell, ArduCoptor 3.5.3 gives no feedback when a polygon fence fails its validity check on load, causing the fence to be non-functional without informing the user at boot, pre-arm, or fence upload. This gives the appearance of an active fence that is not there.

Happened to us with MavProxy 1.5.7 which uploads the fence center point as the return point. With a dog leg shaped fence, the center point(return point) was outside the geofence bounds, which causes the fence validity check to fail.

#### Version
3.5.3

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Octacoptor

#### Hardware type
Pixhawk

#### Logs
NA

OXINARF, 2017-11-01:
Although it looks like the issue exists (no warning/pre-arm check if the polygon fence is invalid), the problem isn't certainly with the return point because that's unused in Copter (and so doesn't count for boundary validity).

Shoob189, 2017-11-01:
I didn't think the return point is used at all in Arducopter, good to have confirmation. I initially believed the return point to be the cause due to the following (checked on branch Copter-3.5):

AC_PolyFence_loader::boundary_valid() checks if the return point is inside the polygon if the contains_return_point parameter is true.

AC_Fence::load_polygon_from_eeprom() calls boundary_valid() with a hard coded true for contains_return_point. The result of the call is assigned to _boundary_valid.

AC_Fence::check_fence() skips over checking the polygon fence if _boundary_valid is false.

OXINARF, 2019-02-01:
Fixed by #7411.

closed 2019-02-01 by OXINARF