ArduPilot/ardupilot #7139: Only the 2nd wheel encoder info is used.

pavloblindnology, 2017-10-27:
#### Issue details
When rotating the encoded wheels individually, only info from the 2nd (WENC2) is used.
Also, the amount of movement is very exaggerated, though settings seems correct.
I use GHM-16 30:1 [motor](http://www.robotshop.com/en/ghm-16-gear-motor-rear-shaft.html) with this [rover](http://www.robotshop.com/en/4wd1-robot-aluminum-kit.html) and 100 CPR [encoders](http://www.robotshop.com/jp/en/lynxmotion-qme-01-quadrature-encoder.html)
So, the settings looks like this
WENC_CPR 100
WENC_RADIUS 0.057
Both encoders report good through rpm1, rpm2 - positive for forward, and similar values.

#### Version
APM:Rover V3.2.0-rc1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ + ] Rover
[  ] Submarine

#### Airframe type
4-wheel rover with 2 encoders.

#### Hardware type
Pixraptor.

#### Logs
_Please provide a link to any relevant logs that show the issue_

pavloblindnology, 2017-10-30:
100 CPR of used encoder correspond to 400 quadrature CPR, which give correct distances.

gmorph, 2017-12-17:
Is this still an issue?

pavloblindnology, 2017-12-18:
This is still an issue. I investigated the reason some time ago, and as I remember it's caused by this [line](https://github.com/ArduPilot/ardupilot/blob/0b78817b816286c8246b2feff4fdaed9a61bd634/libraries/AP_NavEKF3/AP_NavEKF3_Measurements.cpp#L144), where single var `wheelOdmMeasTime_ms` is used for storing the last time update for both wheels, so that one of the wheels always fails the check.

peterbarker, 2021-07-30:
This was resolved by #12578 - thanks

closed 2021-07-30 by peterbarker