ArduPilot/ardupilot #7141: AP_RSSI: Add support for Spektrum Satellite "Fade Counter" to fake RSSI

DavidIngraham, 2017-10-27:
#### Issue details
_Please describe the problem, or desired feature_
Spektrum Satellite Receivers do not support RSSI, but do report a "Fade Counter" that signifies the number of dropped packets. This is documented here: [https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf](https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf)

While it is not a true RSSI value, I would argue that it would still be helpful to support this though the AP_RSSI library. AP_RSSI already supports reciever-protocol specific RSSI, so I think the changes would only need to be implemented in the PX4_firmware layer.

Cleanflight supports this. They have some good discussion here: https://github.com/cleanflight/cleanflight/issues/1658

Cleanflight 
#### Version
_What version was the issue encountered with_
N/A (Enhancement)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
All

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_
N/A

geofrancis, 2017-10-28:
I believe this has been implimented with sbus so rssi = % of packets received.

OXINARF, 2017-11-05:
I'm not closing this as we don't usually do that for feature requests, but doesn't look very useful. Besides the technical difficulty of knowing the format of the received packet, the fades field will have jumps (and so will the RSSI) and when the limit is reached it doesn't even wrap around, it just stays fixed.

geofrancis, 2017-11-05:
This #3274 is the commit that does what is requested here with SBUS and sumd instead of DSM.