ArduPilot/ardupilot #7145: Erroneous Rangefinder values from uLanding Radar Altimeter

Evan1010, 2017-10-28:
#### Issue details
_Diagnosed via Mission Planner 1.3.49 and 1.3.50. SonarRange return initially reads "0". After some time, the SonarRange value is populated: in most cases, a value of 6.35 (cm) is displayed regardless of orientation and obstructions before the altimeter. On occasion, sporadic (erroneous) values are displayed. This problem was originally (Plane 3.8.0) related to the firmware changes made by Aerotenna between uLanding pre-production versions and later revisions. Plane 3.8.2 Implemented updated code in AP_Rangefinder library driver AP_RangeFinder_uLanding.cpp in an attempt to resolve the individual firmware revisions through format detection. This update, however, yields the same behavior. See [here](https://aerotenna.readme.io/v1.0/discuss/599ece977c48d8002381d64a) for issue confirmation from Aerotenna proper, and [here](https://discuss.ardupilot.org/t/aerotenna-ulanding-extensive-issues/20062/4) for personal analysis and my demonstrated working patch._

#### Version
_Plane 3.8.0 through 3.8.2_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_Flying Wing_

#### Hardware type
_Pixhawk 2.1_

#### Logs
_See [here](https://discuss.ardupilot.org/t/aerotenna-ulanding-extensive-issues/20062) for extended diagnostic and temporary patch for current (revision 1) uLanding altimeters._

davidaroyer, 2017-10-30:
It looks like the updated uLanding driver just hasn't made it into the stable releases yet. I'm looking specifically at the release [ArduPlane-3.8.2](https://github.com/ArduPilot/ardupilot/releases/tag/ArduPlane-3.8.2), and that uLanding driver is still the old one that only works with the original uLanding data format.

Evan1010, 2017-10-31:
I replied to the parallel Ardupilot discourse, acknowledging my mistake - thank you for clearing that up.

Should this issue thread be closed, then?

auturgy, 2017-11-03:
Please leave it open until the issue is resolved in a release

magicrub, 2017-11-03:
I've added the DevCallTopic tag to discuss back-porting this to the minor release cycle. Plane currently has v3.8.3beta1 out so maybe we can slip it into there soon for a beta2

Evan1010, 2017-11-03:
Compiled ArduPlane (PX4-v3) from source code in Master last night. 
Verified that the uLanding altimeter does indeed work with the updated driver.

So it should be a matter of time until the next release (alternatively, using 3.8.3 Beta 2, which is already out).

OXINARF, 2017-11-05:
@magicrub @tridge Can this be closed? The driver just needs to be backported if that's wanted, there is no actual issue.

magicrub, 2017-11-05:
Confirmed fixed, closing. I'll about helping Tridge get it into the next minor release

closed 2017-11-05 by magicrub