ArduPilot/ardupilot #7156: Copter: automatic takeoff without using the throttle safety (only for AUTO flight mode)

robustini, 2017-10-31:
#### Issue details
As is known "Ardupilot QuadPlane" is able to takeoff automatically with a Mavlink command without the "throttle safety", using the "VTOL_TAKEOFF" event.
In the Copter this security has always existed, it would be interesting to disable it with an option.
From the wiki:

"If starting the mission while the copter is on the ground the pilot should ensure the throttle is down, then switch to the Auto flight mode, then raise the throttle. The moment that the throttle is raised above zero, the copter will begin the mission"

It would be interesting to disable throttle safety, only during "Auto" mode with a takeoff event.

#### Version
3.5.3

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2017-10-31:
I think in Copter's possible to take-off without the pilot having to raise the throttle if a command-long is sent with either a TAKEOFF or MISSION_START command within it.

robustini, 2017-10-31:
In SITL it doesn't work, I want to arm from the Planner, and choose AUTO, the Copter should start the mission if the first event is "TAKEOFF", and it doesn't happen.

closed 2020-12-16 by robustini

rmackay9, 2020-12-17:
Thanks for closing this.  We have AUTO_OPTIONS in 4.1 which allows this now I think.