ArduPilot/ardupilot #7157: RTL descent check error

RoelHelsen86, 2017-10-31:
#### Issue details
When using RTL to descend until a certain altitude, sometimes the copter does not finish the mission after the descent stage of rtl. I traced this back to altitude check to confirm that the desired altitude is reached. This check has a tolerance of 20cm which is too narrow given the fact that a different altitude is used than the one that copter uses to steer with, and so the altitude can vary from the one the copter is holding. This difference can be larger thatn 20cm.

control_rtl has the following line as completion check
// control_rtl .cpp - line 337
rtl_state_complete = abs(rtl_path.descent_target.alt - current_loc.alt) < 20;
while the copter uses inav.get_altitude() for control and logging.

I cannot provide logs since as far as I found, current_loc.alt is not logged anywhere. However, it is sent via mavlink. On one flight I had current_loc.alt around 20.6m while the log shows that it was at 20m sharp. Because of this discrepancy, the copter did not consider the rtl complete.

My main questions are:
1) why does control_rtl not use inav.get_altitude()?
2) why is there the discrepancy between current_loc.alt and inav.get_altitude()?

I post this here, as I consider it an issue of robustness in the code. The implications of this might be minor in most cases, but in my application it breaks down my system because the copter does not mark rtl as complete. I can also imagine that more people have this problem and that it might happen at more places in the Arducopter code.

#### Version
Copter-3.5.3

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X8

#### Hardware type
pixhawk2

#### Logs
n.a.

rmackay9, 2017-11-01:
The difference is based on the alt-above-ekf-origin (which inav.get_altitude returns) and the alt-above-home (which is what is held in current_loc.alt and the rtl_path variables).

the completion check highlighted in the "rtl_descent_run" method is only used if the RTL_ALT_FINAL parameter is set to non-zero.  Most people don't set this parameter so it's possible we have an issue in the code but I'd like to confirm that you *do* have this parameter set to a non-zero number?

By the way, that "abs" should almost certainly be a "labs".  So we should fix that at least.

RoelHelsen86, 2017-11-02:
Thanks for the quick reply.
Indeed, I have the RTL_ALT_FINAL set to non-zero, because iI want the drone to return home and then do my own landing procedures.

What I saw is that there is a small difference between both altitudes and that the check is too narrow (20 cm) to deal with this discrepancy (60cm in my case). In my opinion the nicest way is to use the same height for control as for the checks, but I understand the difference between both altitudes and why the above home one is used.

Can I just change the abs() to labs() to solve that already?

rmackay9, 2017-11-02:
I don't think changing the "abs()" to "labs()" will resolve the issue (assuming we have an issue).

Can I ask, how do you know, "When using RTL to descend until a certain altitude, sometimes the copter does not finish the mission after the descent stage of rtl.".  Is there a message or something that you're expecting to see somewhere that tells you the mission is complete?

RoelHelsen86, 2017-11-02:
I was not expecting the labs() to solve the issue, but I wanted to use the more correct one anyway

I am expecting a MISSION_ITEM_REACHED (#46) mavlink message when it completes the RTL as i am running RTL as part of an AUTO mission. This has been working quite well, except occasionaly it does not give the MISSION_REACHED message when it finishes the RTL. The last time that this happened I looked a bit closer and discovered that the current_loc.alt was different from the altitude from the logs by more than 20cm, leading me to report the issue.

khancyr, 2021-01-07:
the rtl logic was change and the previous code isn't there anymore.
I cannot replicate the issue anymore.
Therefore, I will close this.
Thanks for reporting, if it is still present on current stable release, feel free to reopen it

closed 2021-01-07 by khancyr

rmackay9, 2021-01-07:
The code is still there actually at the bottom of mode_auto.cpp's descent_run() method. https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_rtl.cpp#L345.

We have changed the "abs()" to a "labs()".