ArduPilot/ardupilot #7159: AP_Camera only send feedback message if autopilot is triggering the camera directly

jmachuca77, 2017-11-01:
#### Issue details
Currently every time the autopilot sends a trigger message the feedback message is also sent. By the pixhawk. In cases where the camera is actually triggered by a secondary camera controller or a companion computer. Then that component should be responsible for sending the feedback message. Only in the case where there is no other component to trigger the camera, or if the feedback pin is used on the pixhawk, should the feedback message be sent.

I propose a parameter to enable or disable sending the feedback message.

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All Airframes

#### Hardware type
Not relevant to hardware

rmackay9, 2017-11-01:
I think maybe a CAMERA_TYPE parameter would be good.  I think we will need different drivers for different cameras some time in the next year.

WickedShell, 2017-11-03:
Also worth noting that if a feedback pin has been setup on the vehicle then I'd actually rather get a feedback message when we see a feedback event rather then continuously. (The feedback message needs some MAVLink2 extensions though to be more robust/useful)