ArduPilot/ardupilot #7166: No wheel odometry

pavloblindnology, 2017-11-02:
#### Issue details
After recent upgrade from APM:Rover V3.2.0-rc1 to APM:Rover V3.2.0-rc2 firmware I cannot see any reaction to wheels rotation, though rpm-s are still reported correctly.
For APM:Rover V3.2.0-rc1 (with the same settings) the position was reacting according to wheel rotation and heading.
Now the position works as if no wheel encoders are installed at all.
Also, after the upgrade, the LOCAL_POSITION_NED stopped being reported.
Also cannot set home_position either through APM Planner nor by "mavros mavcmd sethome".

#### Version
APM:Rover V3.2.0-rc2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ + ] Rover
[  ] Submarine

#### Airframe type

#### Hardware type
Pixraptor

#### Logs
Messages from GCS and [tlog](https://drive.google.com/open?id=0B7OhM4lSIhQQTi1HbE9mVGVvOFE)
[MAV 001:1] APM:Rover V3.2.0-rc2 (22fe334e) 
[MAV 001:1] PX4: 0384802e NuttX: 1bcae90b 
[MAV 001:1] PX4v3 002A002F 30365101 31323731 
[MAV 001:1] APM:Rover V3.2.0-rc2 (22fe334e) 
[MAV 001:1] PX4: 0384802e NuttX: 1bcae90b 
[MAV 001:1] PX4v3 002A002F 30365101 31323731 
[MAV 001:1] EKF3 IMU0 buffers, IMU=19 , OBS=7 , dt=0.0200 
[MAV 001:1] EKF3 IMU1 buffers, IMU=19 , OBS=7 , dt=0.0200 
[MAV 001:1] EKF3 IMU0 initialised 
[MAV 001:1] EKF3 IMU1 initialised 
[MAV 001:1] EKF3 IMU0 initial yaw alignment complete 
[MAV 001:1] EKF3 IMU1 initial yaw alignment complete 
[MAV 001:1] EKF3 IMU0 tilt alignment complete 
[MAV 001:1] EKF3 IMU1 tilt alignment complete 
[MAV 001:1] GPS 1: detected as u-blox at 115200 baud 
[MAV 001:1] u-blox 1 HW: 00080000 SW: 2.01 (75331)

gmorph, 2017-12-17:
Could you please attach a dataflash log file with the issue?  Thanks.

pavloblindnology, 2017-12-18:
On RC1 wheels worked again.
I'll prepare the log file when I have time.
Thanks.

peterbarker, 2020-12-07:
We're quite sure wheel odometry works now in current versions - we had a GSoC student base BalanceBot support around them between your report here and now.

Thanks for the report!

closed 2020-12-07 by peterbarker