ArduPilot/ardupilot #7169: "MAG_ENABLE" Disable.

liang-tang, 2017-11-03:
#### Issue details
I found a question that is if I set MAG_ENABLE to 1, I can't arm the vehicle.
#### Version
Copter 3.5.3

#### Platform
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

auturgy, 2017-11-04:
Please post support requests to discuss.ardupilot.org

rmackay9, 2017-11-04:
It's a duplicate of this existing issue so closing.
https://github.com/ArduPilot/ardupilot/issues/5549

closed 2017-11-04 by rmackay9