ArduPilot/ardupilot #7170: Add multiple takeoff and landing capability without the use of a companion computer

frizensami, 2017-11-03:
#### Issue details
As discussed with @tridge in [this forum post](https://discuss.ardupilot.org/t/anyone-has-experience-with-long-flights-in-unfamiliar-terrain/22410/8), flying a mission with multiple takeoffs and landings requires something like a companion computer to run the rest of the mission after landing for the first time if no direct telemetry control is present (i.e. re-arm the aircraft after landing, switch to next waypoint). 

Removing the need for a companion computer in this situation will be helpful for an increasingly common use case now: autonomous delivery vehicles that operate BVLOS and out of telemetry range and need to
* Land at a remote site and upon cargo loading/unloading and on a button press, take off and RTL
* Or conduct another leg of the mission to another area, doing multiple deliveries in on a single battery/fuel load

Also as discussed, this could be done with an addition to `AP_Button` such as `BTN_ACTION1` or `BTN_ACTION2`, of which one option could be `1: Arm and switch to next mission item`. 

I currently would like this functionality on a cargo delivery aircraft being built that is running Ardupilot, and since it will run in a rugged/rural environment, I would prefer having fewer components on board to service to make the entire system more reliable

#### Version
N/A

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[X] Submarine

#### Airframe type
X-Frame quadplane currently, but this is more widely applicable

#### Hardware type
Pixhawk 2.1 (Cube), ditto

#### Logs
N/A

rmackay9, 2017-11-03:
I think this request applies mostly to Plane so I've added "Plane:" to the subject line.  I think copter does allow a land and then a take-off within the same mission although I haven't tried that in a while and there might be some small issue stopping this from working.

magicrub, 2017-11-03:
I am currently (but slowly) working on a TAKEOFF flight mode which will help address this a little. Also,  about a year ago @WickedShell and I split fixed-wing landing into it's own library to help make more advanced landing options with this concept in mind. So, these things are in the works but the details are not quite worked out so I'll leave this issue open for discussion.

tridge, 2017-11-04:
randy, the key is to have a button on the aircraft (not a TX button) which allows for someone near the aircraft to tell it to arm and continue the mission
I don't think there is any way to do that in copter. Thus the suggestion to use the BTN_* parameters

WickedShell, 2017-11-04:
Notionally I'd be quite happy with hooking the safety button to also do a mission restart when it's activated, Plane does support at least normal fixed wing takeoffs waiting until the button has been depressed first (but it must already be on the takeoff item).

If people are fine with an optional restart on button activation I'm happy to send the PR in for it.

tridge, 2017-11-04:
i don't like it being the safety button as that would also be used when prepping the plane, and I think pressing safety and it then starting the motors for a mission would be very bad

WickedShell, 2017-11-04:
Yeah, and I agree with that overall. I was mostly observing that plane supports this reasonably well on fixed wing. Quadplane however has a lot of problems, but on a normal fixed wing aircraft this can be quite safe.