ArduPilot/ardupilot #7176: Visual and wheel odometry cannot be used together

pavloblindnology, 2017-11-03:
Do I understand right, that currently, due to the following lines, only either visual or wheel odometry can be fused, but not both?

https://github.com/ArduPilot/ardupilot/blob/e63e39c411f463c755e02f137dce39c255633f04/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L1566

https://github.com/ArduPilot/ardupilot/blob/e63e39c411f463c755e02f137dce39c255633f04/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L1579

priseborough, 2017-11-03:
Not on the same frame, however if wheel odometry data is available in the buffer it should be picked up on the next frame, but will be fused using states that are 10msec older which is not a problem in practice.

pavloblindnology, 2017-11-06:
Thanks for the explanation.

closed 2017-11-06 by pavloblindnology