ArduPilot/ardupilot #7178: SITL build fails, fatal error no function "pthread_setname_np"

guglie, 2017-11-04:
#### Issue details
SITL build with sim_vehicle.py fails

I'm using this command:
`sim_vehicle.py -v ArduCopter -f quad --no-mavproxy`

#### Version
Latest version on this repository, https://github.com/ArduPilot/ardupilot/commit/c9652fec3ccd957e1082dc9f1f27ea7bd419be9f

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter *(maybe others too?)*
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quad *(maybe others too?)*

#### Hardware type
None, SITL

#### Logs

```
SIM_VEHICLE: Building
SIM_VEHICLE: "ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `ardupilot/build/sitl'
[458/828] Compiling libraries/AP_AccelCal/AP_AccelCal.cpp
[467/828] Compiling libraries/SITL/SIM_FlightAxis.cpp
../../libraries/SITL/SIM_FlightAxis.cpp:93:5: fatal error: no matching function for call to 'pthread_setname_np'
    pthread_setname_np(pthread_self(), "ardupilot-flightaxis");
    ^~~~~~~~~~~~~~~~~~
/usr/include/pthread.h:475:5: note: candidate function not viable: requires 1 argument, but 2 were provided
int     pthread_setname_np(const char*);
        ^
1 error generated.

[825/828] Compiling libraries/SITL/SIM_Plane.cpp
[826/828] Compiling libraries/SITL/SIM_Sprayer.cpp
Waf: Leaving directory `ardupilot/build/sitl'
Build failed
 -> task in 'objs/SITL' failed (exit status 1): 
	{task 4568049536: cxx SIM_FlightAxis.cpp -> SIM_FlightAxis.cpp.0.o}
['/usr/bin/clang++', '-std=gnu++11', '-fdata-sections', '-ffunction-sections', '-fno-exceptions', '-fsigned-char', '-Wall', '-Wextra', '-Wformat', '-Wshadow', '-Wpointer-arith', '-Wcast-align', '-Wundef', '-Wno-unused-parameter', '-Wno-missing-field-initializers', '-Wno-reorder', '-Wno-redundant-decls', '-Wno-unknown-pragmas', '-Werror=format-security', '-Werror=array-bounds', '-Werror=uninitialized', '-Werror=init-self', '-Werror=switch', '-Wfatal-errors', '-fcolor-diagnostics', '-Wno-gnu-designator', '-Wno-inconsistent-missing-override', '-Wno-mismatched-tags', '-Wno-gnu-variable-sized-type-not-at-end', '-Wno-c++11-narrowing', '-O3', '-include', 'ap_config.h', '-Ilibraries', '-Ilibraries/GCS_MAVLink', '-I.', '-I../../libraries', '-I../../libraries/AP_Common/missing', '-DSKETCHBOOK="ardupilot"', '-DCONFIG_HAL_BOARD=HAL_BOARD_SITL', '-DCONFIG_HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE', '../../libraries/SITL/SIM_FlightAxis.cpp', '-c', '-oardupilot/build/sitl/libraries/SITL/SIM_FlightAxis.cpp.0.o']
SIM_VEHICLE: Build failed
```

peterbarker, 2017-11-05:
On Sat, 4 Nov 2017, guglie wrote:

> /usr/include/pthread.h:475:5: note: candidate function not viable: requires 
> 1 argument, but 2 were provided
> int     pthread_setname_np(const char*);
>         ^
> 1 error generated.

This code was recently changed, but works appropriately on Linux.

What platform are you running on?

guglie, 2017-11-06:
Mac OSX, I tried removing the first argument `pthread_self(), ` as pointed by @OXINARF and it builds and runs.

PR #7337 (2017-12-04): SITL: fix pthread_setname_np to build on mac
This fixes #7178 (SITL build failing)
closed 2017-12-04 by peterbarker