ArduPilot/ardupilot #7192: Allow precland to work indoor

expntly, 2017-11-07:
#### Issue details
I tried precland (companion board) indoor, with GPS disabled (GPS_TYPE=0, AHRS_GPS_USE=0, EK3_GPS_TYPE=3) and a lidar, but it doesn't pick up on the target.  PL.TAcq shows 0, and further digging seems to indicate this is the problem: https://github.com/ArduPilot/ardupilot/blob/6a94d285de43de976e28bbab4df725c13fd96591/libraries/AC_PrecLand/AC_PrecLand.cpp#L233

https://github.com/ArduPilot/ardupilot/blob/6a94d285de43de976e28bbab4df725c13fd96591/libraries/AP_NavEKF2/AP_NavEKF2_Control.cpp#L490

It would be great to be able to fly indoor for testing, and let the camera override.

@rmackay9 @jschall 

#### Version
Head

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X4

#### Hardware type
Pixhawk2.1

OXINARF, 2017-11-07:
@jschall can say more, but I think that not using velocity would be a regression and quite likely wouldn't be very useful for indoors.

expntly, 2017-11-07:
I think it would be useful to test precland indoor.

Is the only way to do this right now using opticalflow and disable GPS?

OXINARF, 2017-11-08:
Using beacons should also work.

When I said that without velocity information, it won't be very useful for indoors is that it will probably overshoot much more and when inside you typically have less space than outside.

expntly, 2017-11-11:
Yeah but the idea is that computer vision should provide you with relatively highly accurate information as to what your positioning is.  I'm happy to test it.  Is there a way to make this work though?  I've tried disabling a few things and a bunch of parameters but it looks like I'm not on the right path.

fnoop, 2018-08-06:
Precland absolutely does work indoors, @patrickpoirier51 has cheated death many times in his garage/mancave to prove it :)

patrickpoirier51, 2018-08-06:
Yep... welcome to the ThunderDrone
Flying in GPS denied requires a mean to estimate position, as an option  the OpticalFlow can be used for this purpose. I used EK3 because you can set the EKF home position  and work in guided mode as well.

For tracking I used, RPI, ODROID with code inspired by @jschall  and tested @fnoop  vision position. It worked as well with OpenMV and JEVOIS camera systems.

peterbarker, 2021-07-30:
Closing as "does work".

rmackay9, 2021-07-30:
I've always interpreted this as meaning, "allowing precision landing target to be used as a position source to the EKF".  I think I'll re-open this because it's a valid enhancement request still I think.

patrickpoirier51, 2021-07-30:
Something like using  precland message to feed vision_position_estimate ?

rmackay9, 2021-07-31:
@patrickpoirier51, yes, something like that.  @rishabsingh3003 and I have discussed this as part of his work on precision landing and he's actually already done some experiments and found that at least with IRLock the position estimate is not very good.  Perhaps with another sensor it might be more useful.