ArduPilot/ardupilot #7206: Rover: Different heading for throttle forward\reverse and arm\dearm.

pavloblindnology, 2017-11-08:
#### Issue details
When armed, heading behaves differently depending on throttle forward or reverse.
If even a little forward then heading is computed in one way, which is not very coincident with compass (sometimes the heading even goes crazy),
if backward - then heading immediately jumps to another state, which coinсides with the compass heading.
The same when armed\disarmed: armed corresponds to forward, disarmed to backward.
The next graphs from APM Planner below corresponds to case 1 [throttle](https://drive.google.com/open?id=1335FoJ4eV_uhVSby-qalY9sJ1-pEV89Y) (green - yaw, purple - throttle) and case 2 [arming](https://drive.google.com/open?id=1TkngdxTC2YiZzzT9AlK2WOp4yHidXkMZ) (green - yaw, jumps corresponds to arm\disarm).

#### Version
APMRover v3.2.0-dev, v3.2.0-rc2, v3.2.0-rc1, v3.1.2.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ + ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
Pixraptor

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-11-08:
@pavloblindnology Those links are not accessible, it asks for authorization.

rmackay9, 2017-11-09:
My guess is this is an EKF type thing so probably one for @priseborough to look at.

pavloblindnology, 2017-11-09:
Updated the links. Sorry.

pavloblindnology, 2017-11-09:
Also, the AHRS system shows unhealthy message when armed and throttle neutral or forward.

gmorph, 2017-12-17:
This is not behaviour I've seen.  You should wait the 10 seconds or so on start up for the AHRS to become Healthy after the GPS lock etc.  If something is unhealthy and your force arming and driving then expect weird behaviour.
Compass is really only used for heading on initial startup of Rover.  Once the rover is driving compass is just one of the inputs. Is your compass affected by your motors?  Are you using an external compass - normally people have one in the GPS?  I know nothing about what a pixraptor hardware is like so if its onboard compass anything could be happening.
We really need a dataflash log file for when the issue occurs to analyse further.  Without that there isn't much we can do.  I'm going to close this for now but if you attach a log file to this issue I'll get a notification and I'll reopen it and investigate.
Thanks, Grant.

closed 2017-12-17 by gmorph

pavloblindnology, 2017-12-18:
I use GPS+Compass external module, so compass isn't affected by the motors.
And, yes, I wait for AHRS to initialize.
I'll prepare the log file when I have time.
Thanks.

rmackay9, 2018-01-23:
Trying to use a GPS indoors is likely related to the issue you're seeing.  With Rover-3.2.1 (entering beta testing soon) you should be able to turn off the GPS and do indoor navigation with just the wheel encoders (and IMU + Compass).

rmackay9, 2018-01-23:
Ah, I forgot to mention, maybe it would be best to move this discussion into the Rover-3.2 forum until we've confirmed it's a bug.  https://discuss.ardupilot.org/c/ardurover/rover-3-2

pavloblindnology, 2018-01-24:
Ok, thanks. I'll open the topic there.
Just want to mention that even though indoor use can be related to the behavior, but still, there is some inherent difference between throttle forward vs throttle backward AHRS computations. And bad GPS seems just reveal it.

rmackay9, 2018-01-24:
@pavloblindnology,
It's very possible that there's an issue in the EKF when we change from "fly_forward" to non-"fly_forward".

This is where we set the ahrs flag: https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/APMrover2.cpp#L172
These are a couple of places where we set the in_reverse flag:
 - https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/system.cpp#L205
- https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode_manual.cpp#L14

I'll ask @priseborough if he has any ideas.

pavloblindnology, 2018-01-24:
@rmackay9 That's the exact place. Thank you for pointing out.

priseborough, 2018-02-20:
I'm having trouble opening that log file with both MAVExplorer and FlightPlot. Looking at EKF3, _ahrs->get_fly_forward is consumed here:https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_Control.cpp#L467

assume_zero_sideslip will always be false if _ahrs->get_vehicle_class() == AHRS_VEHICLE_GROUND

pavloblindnology, 2018-02-20:
I can open the log in MissionPlanner but not in APMPlanner.
fly_forward flag is also consumed in AP_AHRS_DCM.cpp and AP_AHRS_NavEKF.cpp, for example
here: https://github.com/ArduPilot/ardupilot/blob/9419cc3b32cceb5a72f9be5799d113d6821ddbeb/libraries/AP_AHRS/AP_AHRS_DCM.cpp#L394
As I understand, it switches between GPS and compass heading, depending on this flag.

pavloblindnology, 2018-02-20:
Yes, you're right. But here, for example, it can fall back to DCM depending on direction in case of no GPS.
https://github.com/ArduPilot/ardupilot/blob/9419cc3b32cceb5a72f9be5799d113d6821ddbeb/libraries/AP_AHRS/AP_AHRS_NavEKF.cpp#L981

peterbarker, 2018-02-20:
On Mon, 19 Feb 2018, Paul Riseborough wrote:

> I'm having trouble opening that log file with both MAVExplorer and FlightPlot. Looking at EKF3, _ahrs->get_fly_forward is consumed

Opens OK with MAVExplorer.py here.  What error are you getting?

priseborough, 2018-02-20:
We need to change that behaviour for rover. It makes no sense to do that. It should always use the same primary source for attitude unless the source declares itself as unhealthy.

priseborough, 2018-02-20:
@peterbarker It looks like it fails to load on my OSX environment, but does load on the ubuntu VM. The error i am getting is: 
[image]

The same error does not occur when I load a plane log from last weekend.

priseborough, 2018-02-20:
FlightPlot generates a different error with that log: 
[image]

priseborough, 2018-02-20:
The MAVExplorer in ubuntu completes the load, but also raises an error message for the ISBD message:

[image]

peterbarker, 2018-02-20:
@priseborough OK, all of those have the same root cause; parsers not understanding new format characters.  Some of them are not handling that gracefully.

The Ubuntu sample you've got is it handling it gracefully.  If you update pymavlink the error will go away (and you can use `mavfft_isb.py` to analyze ISBH/ISBD messages (fast-rate sampling occurs for a short period of time, the data is then dribbled out to the log).  You get fft-able data with minimal performance and logging-size impact).

The MacOS thing is the same thing but the error isn't being handled gracefully.  Is that some truly ancient version of pymavlink?

The FlightPlot thing is just bizarre.  Not understanding `d` is weird - it's "double", and not supporting doubles in logs is strange.  Of course, if it doesn't understand `d`, the chances of it understanding `a` are low, so it is almost certainly doubly-stuffed.  The really bad thing is it not skipping over stuff it doesn't understand; the standard "liberal in what you accept" thing.

priseborough, 2018-02-20:
Thanks @peterbarker. I did a pymavlink update on both Ubuntu and OSX and the error has disappeared. Python version was 2.7.10

rmackay9, 2018-02-22:
I've created a new issue here: https://github.com/ArduPilot/ardupilot/issues/7760