ArduPilot/ardupilot #7207: MAVLink: send_to_components(&msg) fails with CRC error for MAVLink 2 msg

SamuelDudley, 2017-11-08:
#### Issue details
During ArduPilot development I noted that MAVLink 2 messages sent via GCS::send_to_components() fail to be decoded correctly on the receiving end **only** when they have been sent from hardware.
For example I have added the following code into AP_Camera:
```
// send camera feedback to message to components of this system
void AP_Camera::send_feedback_ahrs() {

    mavlink_message_t msg;
    mavlink_camera_feedback_ahrs_t camera_feedback_ahrs = { };

    camera_feedback_ahrs.img_idx = _image_index;
    camera_feedback_ahrs.trigger_time = _camera_feedback_time;
    camera_feedback_ahrs.sample_time = _ahrs_summary.ahrs_update_time;
    camera_feedback_ahrs.x = _ahrs_summary.ned_pos_rel_home.x;
    camera_feedback_ahrs.y = _ahrs_summary.ned_pos_rel_home.y;
    camera_feedback_ahrs.z = _ahrs_summary.ned_pos_rel_home.z;
    camera_feedback_ahrs.vx = _ahrs_summary.velocity.x;
    camera_feedback_ahrs.vy = _ahrs_summary.velocity.y;
    camera_feedback_ahrs.vz = _ahrs_summary.velocity.z;
    camera_feedback_ahrs.q1 = _ahrs_summary.quat.q1;
    camera_feedback_ahrs.q2 = _ahrs_summary.quat.q2;
    camera_feedback_ahrs.q3 = _ahrs_summary.quat.q3;
    camera_feedback_ahrs.q4 = _ahrs_summary.quat.q4;
    camera_feedback_ahrs.home_lat = _ahrs_summary.home.lat;
    camera_feedback_ahrs.home_lon = _ahrs_summary.home.lng;
    camera_feedback_ahrs.home_alt = _ahrs_summary.home.alt;
    camera_feedback_ahrs.lat = _ahrs_summary.location.lat;
    camera_feedback_ahrs.lon = _ahrs_summary.location.lng;
    camera_feedback_ahrs.alt = _ahrs_summary.location.alt;
    camera_feedback_ahrs.target_component = _vision_feedback_target_component;

    // encode camera feedback ahrs into MAVLink msg
    mavlink_msg_camera_feedback_ahrs_encode(0, 0, &msg, &camera_feedback_ahrs);

    // send to all components
    GCS_MAVLINK::send_to_components(&msg);
}
```

If I define ` camera_feedback_ahrs` as a MAVLink 2 message and flash the code onto hardware, `camera_feedback_ahrs` messages will appear as `BAD_DATA` after failing a CRC check (pyMAVLink). Note that other MAVLink 2 messages are received and parsed fine by the same message definitions leading me to believe that the CRC error is unique to messages sent via the `GCS_MAVLINK::send_to_components( )` static method.

- This error does **not** occur when running the ArduPilot code in SITL (messages are received correctly irrespective of the MAVLink version)
- This error **only** occurs when the message is defined as a MAVLink 2 message (the same message definition as a MAVLink 1 message works on both hardware and SITL)

#### Version
Current master

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
N/A

#### Hardware type
PX4-V3 / the cube

#### Logs
N/A

SamuelDudley, 2017-11-09:
I can confirm that the `TEST` message defined above is received correctly using this sending method from the cube:
```
    mavlink_msg_test_send(chan,
        AP_HAL::micros64(),
        3.0,
        4.0);
```
decoded result:
```
TEST {time_usec : 50500552, field1 : 3.0, field2 : 4.0}
```

While the the same message fails to be decoded using the send to components method from the cube:
```
    GCS_MAVLINK::send_to_components(&msg);

    mavlink_message_t msg;
    mavlink_test_t test = { };
    test.time_usec = AP_HAL::micros64();
    test.field1 = 1.0;
    test.field2 = 2.0;

    mavlink_msg_test_encode(0, 0, &msg, &test);
    GCS_MAVLINK::send_to_components(&msg);
```
Decoded result:
```
BAD_DATA {unknown MAVLink message ID 13, data:['fe', '10', '5', '0', '0', 'd', '39', '30', '0', '0', '0', '0', '0', '0', '0', '0', '80', '3f', '0', '0', '0', '40', '28', '50']}
```

OXINARF, 2017-11-09:
I think the issue is that `send_to_components` is acting like it is re-sending a message, but, in reality, this method is used to send new messages to just components of the system (vs broadcast in all MAVLink ports). That means it doesn't finalize the message, but instead tries to send it as is.

SamuelDudley, 2017-11-09:
Glad it's not just me! Based on a quick dig into the different paths the messages take to be sent I will agree with your comment. Unfortunately I won't be of any further help here, but I'm happy to test a patch once it becomes available. Thanks.

magicrub, 2017-11-09:
What is the expected behavior? How is send_to_components() different than any given _send() function?

SamuelDudley, 2017-11-09:
My understanding is that `send_to_components( )` allows messages to be sent on channels which have been identified to have a component of that system attached to it. For example I'm sending a message at ~30Hz to a companion computer using `send_to_components( )` to avoid flooding the GCS telemetry channel with unnecessary data. As another example it's also used in `AP_Camera` to send trigger events to attached MAVLink enabled cameras.

From my point of view working behaviour would allow me to send a valid MAVLink 2 message using `send_to_components( )` in the same way it is currently possible with a MAVLink 1 message.

tridge, 2017-11-12:
@SamuelDudley 
I've done a commit here to fix this:
https://github.com/tridge/ardupilot/commits/pr-send-components
I haven't tested it though, if you could please test it before we put it in master that would be appreciated

SamuelDudley, 2017-11-13:
Thanks, doing a drive from VIC to SA today but will test tomorrow.

SamuelDudley, 2017-11-14:
@tridge With the patch applied MAVLink2 messages are now received and decoded via `send_to_components()` when using hardware, however the values of the message fields are incorrect for both SITL and hardware:

```
    mavlink_msg_test_send(chan,
        AP_HAL::micros64(),
        3.0,
        4.0);
```
results in (for both SITL and hardware)
```
TEST {time_usec : 338018226, field1 : 3.0, field2 : 4.0}
```
So that is still fine as per before, however
```
    mavlink_message_t msg;
    mavlink_test_t test = { };
    test.time_usec = 12345;
    test.field1 = 1.0;
    test.field2 = 2.0;

    mavlink_msg_test_encode(0, 0, &msg, &test);
    GCS_MAVLINK::send_to_components(&msg);
```
results in (for both SITL and hardware)
```
TEST {time_usec : 281475261828256, field1 : 3.95394858849e-39, field2 : 1.72990295421e-41}
```
So message framing is seems correct, but the payload is invalid when using `send_to_components()`.

peterbarker, 2021-07-30:
I think I may have fixed this with
```
commit a96a4c05101d637c0e25cb6d70634a193075e73b
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Thu Oct 17 18:07:23 2019 +1100

    GCS_MAVLink: correct encoding used for send_to_components
    
    These have to be packed onto the channel - otherwise they may ge
    tencoded as mavlink1 instead of mavlink2 (or vice-versa)
```

Please reopen if that's incorrect.

closed 2021-07-30 by peterbarker