ArduPilot/ardupilot #7221: Change the MAVLink #WIND message output to 0-360

fredrikfalkman, 2017-11-13:
Noticing that MAVLink spec doesn't prescribe how to present wind direction:

http://mavlink.org/messages/ardupilotmega#WIND

I think that it would make more sense for most pilots to get the wind direction in 0-360 degrees rather than the current -/+ 180 degrees.

Here's where it comes from:
https://github.com/ArduPilot/ardupilot/blob/8178ab4037b05c515946f1f3fe501babaf48ad5b/ArduPlane/GCS_Mavlink.cpp#L296-L305

#### Platform
[ X] Plane (?)

WickedShell, 2017-11-13:
This would break how every ground station expects to interact with the wind message, and for a pilot's interpretation of the wind it is straightforward for a ground station to adjust how it displays the wind.

closed 2017-11-13 by WickedShell

fredrikfalkman, 2017-11-13:
@WickedShell so you think this should be handled on the GCS side? Fine by me.

Got this answer from @DonLakeFlyer on Gitter (Nov 10 18:30):

> For ArduPlane the wind value comes straight from the WIND mavlink message. If it's negative then the firmware sent a negative value.

DonLakeFlyer, 2017-11-13:
> This would break how every ground station expects to interact with the wind message

Huh? Why?

DonLakeFlyer, 2017-11-13:
Is this wind direction relative to the vehicle flight direction?

fredrikfalkman, 2017-11-13:
I always figured - an it definitely seems when flying - that it’s showing true North +/-180 degrees.

WickedShell, 2017-11-13:
> Huh? Why?
Because GCS's have to store/use this somewhere for rotation/display of assets. Changing the mapping of the input can cause a GCS to arrive at different results. (Wind should be a bit better because it is a wrapping one, but in general changing any unit sent to the GCS's has been rejected. (Even when the ArduPilot impl is just wrong to the MAVLink spec).

Wind direction is relative to true north.

fredrikfalkman, 2017-11-14:
So @DonLakeFlyer is the consensus here that this should be dealt with on the GCS side after all? I'd be happy to open an issue!

DonLakeFlyer, 2017-11-14:
Yeah, open a QGC issue on it