ArduPilot/ardupilot #7227: Loiter turns - doesn't work correctly yaw and define degrees of command

Skalaw, 2017-11-14:
#### Issue details
Hello.
I'm in the process of creating a functionality using cmd MAV_CMD_NAV_LOITER_TURNS.
I would like implement 2 features, but I cannot get them working.
First one, during the execution of this command, I wanted yaw to keep moving in the direction of the flight.

> Mission Param #4 | Forward moving aircraft this sets exit xtrack location: 0 for center of loiter wp, 1 for exit location. Else, this is desired yaw angle |

Did not work, yaw is always directed towards the center of loiter wp

Second functionality, I would like to be able to define degrees of the loiter. For example when i set param1 (turns) 0.5f ( 180°) should fly only half of the circle.
I would appreciate all help regarding those issue. Could someone confirm to me either those functionalities are possible to implement with the latest Arducopter?

#### Version - Copter 3-5
#### Platform - Copter
#### Hardware type - pixhawk

amilcarlucas, 2017-11-14:
https://github.com/ArduPilot/ardupilot/pull/5518

and

https://github.com/ArduPilot/ardupilot/pull/5507

Might help you geting started.

OXINARF, 2017-11-17:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

For your first point: that description is from the MAVLink spec that doesn't necessarily match ArduPilot implementation. Param 4 isn't supported in Copter.
For your second point: no, there is no way to do that, param 1 should be an integer.

closed 2017-11-17 by OXINARF