ArduPilot/ardupilot #7228: Problems with building example sketches on Windows

Kamil8, 2017-11-14:
#### Issue details
System can't use waf commands. 
[image]

Dev documentation didn't help.

For me only solution was change operating system to ubuntu. There was no problem to build it, except missing compiler for pixhawk.

auturgy, 2017-11-15:
Need more details please.
Operating system? which instructions did you follow?

auturgy, 2017-11-24:
Waf with Windows is a known issue. You can use WSL, to use waf, or build with make. Instructions for both are in the wiki.  If you have suggestions to make the wiki clearer, please submit a pr :)
Thanks.

OXINARF, 2017-12-18:
Closing this after more than one month without any answer.

closed 2017-12-18 by OXINARF