ArduPilot/ardupilot #7246: AP STorM32 mount and STorM32 controller pitch&yaw reversal issue

olliw42, 2017-11-19:
In a discussion here https://discuss.ardupilot.org/t/why-does-pitch-get-reversed-for-storm32-gimbals/22871 it has been found that the STorM32 Mavlink mount (MNT_TYPE = 4) and the STorM32 gimbal controller are affected by a longstanding bug, which shows up only in MAVLINK_TARGETING mode.

The following text is going to be a bit longer than what one may consider acceptable, since, while the problem itself and it's canonical solution are simple, finding the best solution may need some discussion. I hence aim at also providing sufficient background info.

This bug is probably responsible for problems with the STorM32 in Solo smart shoots. There is no confirmation yet that what follows does in fact resolve these problems, but the bug I describe appears so clear that I think rasing an issue is waranted.

The bug:
--------

Fundamentally, the issue is that the STorM32 internally uses a different coordinate and Euler angle system than ArduPilot. This, when allowing some inaccuracies, boils down to that the signs for pitch and yaw need to be reversed. 

In hindsight it is not clear how this bug ever could have made it into both the STorM32 and the STorM32 mount codes, but it goes years back to when  the Mavlink support was added to the STorM32 and the STorM32 mount was created. The required sign changes for pitch and yaw had been noted very early on, but handling it has been done wrongly, and it remained so since then.

Currently, the required sign changes are done in ArduPilot by the STorM32 Mavlink mount, see here
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L138-L140

This code part is always called, EXCEPT when the mount is in MAV_MOUNT_MODE_MAVLINK_TARGETING, see here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L50-L53

Thus, whenever the gimbal shall be controlled in MAV_MOUNT_MODE_MAVLINK_TARGETING mode, e.g. via an external Mavlink message, when the sign change is not done and the gimbal will response incorrectly, i.e. with reversed pitch and yaw.

The canonical solution:
-----------------------

The obviously correct approach should be that the sign changes are done within the STorM32 gimbal controller. This would ensure that any Mavlink MOUNT_CONTROL message results in the same effect, independent on which party has emitted it or where it came from. It also just makes logical sense.

Accordingly, in the next release the STorM32 firmware will be changed such as to do the sign reversal.

This would mean that the code for the STorM32 Mavlink mount needs to ve changed too, that is that this code piece
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L138-L140
is deleted.

This would resolve that.

I'd like to add that this would also remove any STorM32-stuff from the STorM32 Mavlink mount, and turn it into what might be considered a generic Mavlink gimbal mount. One may want to rename things accordingly. This corrected mount might be also usable with AlexMos.

So far, so good, if there weren't potential compatibility issues.

Backwards compatibility:
------------------------

Sadly, the issue is also present in the old I2C-based STorM32 firmware, which is not maintained anymore since ca 1.5 years, but which nevertheless seems to be still in frequent use. In this firmware the issue would not be corrected (as it is not maintained anymore), which means that the above code change in ArduPilot would make these user's setups non-functional. I have no idea how many I2C-STorM32-ArduPilot exist, and how serious this is. 

I want to stress that personally I would not mind that (I have abandoned I2C long ago ;)), and just would do the canonical solution. 

However, I'm not sure how many ArduPilot users this might make unhappy. In order to avoid this happening, one could imagine several potential approaches. E.g., one could leave the "old" STorM32 mount and add a new Mavlink mount. Or one changes the STorM32 mount as described, and adds a "compatibility parameter". Or one just directs all these users to teh STorM32 serial mount. Or ...

As said, I have no opinion on this. I just thought I should mention it.

cheers, Olli

auturgy, 2017-11-19:
Thanks Olli.
I suspect that the majority of ArduPilot/STorM32 users are using the older i2c firmware, as these controllers are widely available and cheap.  I’d hope a 3rd serial mount for STorM32 would be unnecessary, and could be avoided. Do you see any negatives in replacing the native serial protocol with a general mavlink mount, and leaving the current STorM32 mavlink mount as it is?

olliw42, 2017-11-19:
I would not think that I'm the best person to discuss this with. I don't have any I2C setup, neither use any of the official mounts, nor have insights into how many people use what. :).

If I would have seen anything I would have considered a reasonable suggestion, then I would have suggested it :).

The mavlink mount has the disadvantage that it doesn't report the actual gimbal orientation, while the serial mount does. I also think that it would be quite difficult to get this working. Years ago I was adding the possibility that the STorM32 emits a mount_status message, which for the older AP versions ( 3.3?, 3.4?, can't recall, not 3.5 for sure) did work great. But it appears, I think since the Mavlink routing rules were established, that this is broken now. That is, as much as I understand, in MP one can't get the gimbal point when in the AUTOPILOT pane. So, yes, IMHO, replacing a serial mount with a Mavlink mount may have negatives. 

BTW, I was confused a bit by you using "**native** serial mount", since "native" is what I use to denote "my" betacopter mount. I assumed in my answer that you meant the MNT_TYPE=5 mount. 

As much as I understand it, the current serial mount should not have the sign issue, since it's capturing all incoming info in any mount mode. So, one might wonder if - in combination with a STorM32 - there is actually any need for any Mavlink mount. It's not clear what advantage it is providing, so far I only could identify disadvantages. In other words, it could be a viable approach to direct all I2C-STorM32ers to the serial mount.

The least invasive approach would probably be to just add a parameter, with which the sign change in the mount is activated/deactivated.

However, this would just ensure compatibility, but would NOT help with the sign reversal in MAVLINK_TARGETING mode.

Let's recall what the root cause of the issue is: In all modes except MAVLINK_TARGETING the data is handled by the mount, and it thus can send out proper information, while for MAVLINK_TARGETING the mavlink message is bypassing the mount. In order to get the I2C STorM32 also working correctly you would have to catch the bypassing mavlink message and modify its content.

With a serial mount one handles all data exchange, and thus has fundamentally better chances to get it fully working with the I2C based STorM32s.

I was btw evaluating the possibility to create a corrected I2C based firmware v0.97, but that's not easy (British understatement), I doubt it's possible at all.

auturgy, 2017-11-19:
Thanks Olli.

olliw42, 2017-11-19:
@

olliw42, 2017-11-19:
sorry, pressed the wrong button LOL

olliw42, 2017-11-19:
@auturgy  please note the EDIT in the first line of the first post

auturgy, 2017-11-19:
That’s why I left such a short comment!
Will watch for a couple of days and see if it clears up

olliw42, 2017-11-21:
Please note the EDIT2 to the first post.

The described bug is a bug, and not the correct behavior (and in the latest v2.33e STorM32 firmware the bug has been removed on the STorM32's side). 

Resolving it may break with the past, and it's very worthwhile to think about how to keep the older setups going, but I clearly think that we want to work towards removing bugs.

As said, the simplest thing would be to add a new parameter to the storm32 mount class to enabled/disable the mount's internal sign change.

OXINARF, 2017-11-22:
Let's make this cristal clear (since the first post is quite confusing): there was no bug in ArduPilot!

The MAVLink spec, as unfortunately frequently happens, doesn't specify any convention so I can't blame @olliw42 for taking a different approach, but it would have been better if it followed what ArduPilot did (as STorM32 v2.33e apparently does now).

Because STorM32 had a different interpretation of the MAVLink spec, ArduPilot had to follow it and invert the sign. Let me clarify that the following isn't true:

> Currently, the required sign changes are done in ArduPilot by the STorM32 Mavlink mount, see here
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L138-L140
>
> This code part is always called, EXCEPT when the mount is in MAV_MOUNT_MODE_MAVLINK_TARGETING, see here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L50-L53

The sign inversion is definitely called even in MAVLINK_TARGETING mode, as is seen by the `update` method calling `send_do_mount_control` independently of the MOUNT_MODE.

What happens with Solo (and possibly other implementations) is that it sends MOUNT_CONTROL messages to system ID 0, which, as is known, means broadcast. As the sender requests, the message is broadcast to all MAVLink channels, which includes the STorM32 link. Solo follows the ArduPilot convention and doesn't do any sign inversion so STorM32 will get conflicting information from ArduPilot (which will send MOUNT_CONTROL messages with the correct sign) and from Solo directly (with the incorrect sign).

Now, as what to do now that STorM32 firmware has been corrected to follow ArduPilot convention: in my opinion, adding yet another parameter just for this doesn't make sense when the I2C boards can just use the STorM32 serial driver (which should even be better than the MAVLink one). So my solution would be to just remove the sign inversion and add info to the wiki that users with the old I2C boards should use the serial driver.

olliw42, 2017-11-22:
@OXINARF  

thx for your additional thoughts. However, I cannot agree with all of them.

>there was no bug in ArduPilot!

I think you put this to a level which wasn't the intention of anybody. As you pointed out it couldn't be called a bug in STorM32 either. I consider it a failure of all of us involved at this time. I could have seen that, you could have seen that, but we all didn't. You should also note that at this point of development the approach was proper (otherwise any involved party could/would/should have injected). It only broke later then the routing rules were clarified and set up. 

Which leads to:

> The sign inversion is definitely called even in MAVLINK_TARGETING mode, as is seen by the update method calling send_do_mount_control independently of the MOUNT_MODE.

I of course do not disagree with that it is called in "update", but I totally disagree with that it is definitely called in TARGETING mode. You in fact happen to give the argument yourself in the paragraph which follows !! You explicitly confirm that by virtue of the routing mount_status messages can reach the gimbal bypassing the  send_do_mount_control() function, and thus do not have a sign reversal !! Thus:

1) Your statement that send_do_mount_control() is definitely called even in MAVLINK_TARGETING mode is in general incorrect. It depends on the target system and component IDs. 
(Your statement would have been appropriate if and only if you could show (prove) that mount_status messages can reach the gimbal only and only via  send_do_mount_control() , which obviously isn't true.)

2) If you reread what I wrote in the above you'll find that I mentioned exactly this mechanism 

I'm sorry, there is no justification for your objection. You rather explicitly confirm what I was saying.

BTW: The case of bypassing mount_status messages does not occur only for tsystem = 0, but also other realistic tsystem,tcomponent combinations, e.g. (1,67) or (71,67) (which would directly address the gimbal). It's just not correct what you're saying. IMHO.

olliw42, 2017-11-22:
sorry, again pressed the wrong button

OXINARF, 2017-11-22:
@olliw42 

> I think you put this to a level which wasn't the intention of anybody.

Like I said, to me, the first post is completely unclear where the problem was. It wasn't in ArduPilot, that's cristal clear, hence my statement.

> As you pointed out it couldn't be called a bug in STorM32 either.

I didn't say that. I said nobody can blame you because the spec wasn't (and still isn't) clear. Still STorM32 is using a MAVLink message defined by ArduPilot so you should have followed the ArduPilot convention. The fact you can't be blamed doesn't mean it wasn't a bug, it certainly was.

>  you could have seen that

I couldn't, I hadn't seen one line of ArduPilot code when that went in (I barely even knew ArduPilot then) 😉

> You should also note that at this point of development the approach was proper (otherwise any involved party could/would/should have injected). It only broke later then the routing rules were clarified and set up.

I disagree, it was still the wrong approach - using the same message with different meanings should never be done. Just because the bug wasn't hit then, doesn't mean it wasn't a bug.

> I of course do not disagree with that it is called in "update", but I totally disagree with that it is definitely called in TARGETING mode.

How is that happening? Can you give a description of events that leads to `update` not being called when it is in TARGETING mode?
I'll give you mine: the GCS/user/CC/whatever sends a MOUNT_CONTROL message, ArduPilot reads it and saves the wanted angles and sets mode to MAVLINK_TARGETING. Then periodically `update` is called and a MOUNT_CONTROL message is sent to STorM32 from the `send_do_mount_control` method.

> You in fact happen to give the argument yourself in the paragraph which follows !! You explicitly confirm that by virtue of the routing mount_status messages can reach the gimbal bypassing the send_do_mount_control() function

I didn't mention bypassing once in my post... And that's because in the situation I described with Solo there is no bypassing, one copy of the message goes to ArduPilot and one to STorM32. ArduPilot then periodically sends its own MOUNT_CONTROL messages to STorM32.
If bypassing would happen (and it can happen in certain circumstances) then ArduPilot wouldn't have the mount in MAVLINK_TARGETING mode...

> Your statement that send_do_mount_control() is definitely called even in MAVLINK_TARGETING mode is in general incorrect. It depends on the target system and component IDs.

Again, definitely happens. If the target system and component ID is such that the message is completely bypassed by ArduPilot then it wouldn't be in MAVLINK_TARGETING mode. `update` would still be called and a MOUNT_CONTROL message with sign inverted would still be sent.

> BTW: The case of bypassing mount_status messages does not occur only for tsystem = 0, but also other realistic tsystem,tcomponent combinations, e.g. (1,67) or (71,67) (which would directly address the gimbal). It's just not correct what you're saying. IMHO.

In those other cases, the bypassing would happen and ArduPilot wouldn't be in MAVLINK_TARGETING mode. By the way, if the message was always sent with those IDs it would always bypass ArduPilot which means the other modes wouldn't work either.

olliw42, 2017-11-22:
LOL. So ArduPilot knowingly implemented code which violated their own standard. Good to know that this is your accepted practice.

> I couldn't, I hadn't seen one line of ArduPilot code when that went in (I barely even knew ArduPilot then) 😉

I know that, and "you" meant ArduPilot. The fact that you didn't had been around at that time you in fact could consider a reason to be careful with such statement since you don't know nothing about what happened when.

> using the same message with different meanings should never be done. 

the attitude wasn't there at that time, an no inconsistency was existent. It came just much later. But you would only know if you would know what has happened when. It is btw not helpful to base statements on private speculations.

If you would have read what I wrote you actually would find that I said myself that such an inconsistency doesn't make sense. 

> How is that happening? Can you give a description of events that leads to update not being called when it is in TARGETING mode?

Well, I tried, with an explicit example.

You're saying that everything is perfectly fine. So, OK, things are perfectly fine, and there wasn't any need for discussion.

Accepted.

At least now I can intentionally press this close button which had so much attraction to me before. Sorry for having caused any inconvenience to you. Be ensured I regret it.

OXINARF, 2017-11-22:
> LOL. So ArduPilot knowingly implemented code which violated their own standard. Good to know that this is your accepted practice.

Oh yes, we bend standards when the peripherals don't work as they should. We care more about things working than following the standards.

> the attitude wasn't there at that time, an no inconsistency was existent. It came just much later. But you would only know if you would know what has happened when. It is btw not helpful to base statements on private speculations.

Git history doesn't lie. The sign inversion was only added one month after the driver was added, meaning that it was a bug fix and not something designed as such since the beginning. Also, you say that there was no inconsistency at that time but that's not true: ArduPilot received the MOUNT_CONTROL message from GCS with one interpretation and then sent it to STorM32 with another - which stayed in code until now.
You also talked about routing coming later, but when the STorM32 MAVLink driver was committed, the routing rules were already there.

> Well, I tried, with an explicit example.

Where is it? If it is about the bypassing, then I've already wrote many times: if the message is bypassed, then ArduPilot isn't in MAVLINK_TARGETING mode...

> You're saying that everything is perfectly fine. So, OK, things are perfectly fine, and there wasn't any need for discussion.

Where did I say that? I'll quote myself:

> Now, as what to do now that STorM32 firmware has been corrected to follow ArduPilot convention: in my opinion, adding yet another parameter just for this doesn't make sense when the I2C boards can [...]

You had a bug in your STorM32 firmware, which ArduPilot compensated for. You have now fixed the bug meaning ArduPilot needs to do something about it now.

Pedals2Paddles, 2017-11-22:
Can we get back to what needs to be fixed rather than why? I think it will be more productive...

For the storm32 mavlink mount library, the angle reversing needs to be removed to be compatible with the latest Storm32 controller firmware? But this would break older i2c Storm32 controllers?

So what is least undesirable, making a new mount driver called Storm32_i2c, or making a parameter called  `MNT_STORM32_TYPE`.  I suspect the latter will be the simplest. 0 for mavlink i2c, 1 for mavlink serial?

OXINARF, 2017-11-23:
> For the storm32 mavlink mount library, the angle reversing needs to be removed to be compatible with the latest Storm32 controller firmware?

Yes.

> But this would break older i2c Storm32 controllers?

Yes, because @olliw42 won't or can't update those old boards.

> So what is least undesirable, making a new mount driver called Storm32_i2c, or making a parameter called MNT_STORM32_TYPE. I suspect the latter will be the simplest. 0 for mavlink i2c, 1 for mavlink serial?

It's not MAVLink I2C, there is no such thing. The STorM32 I2C boards were boards where components were connected via I2C, newer ones use a different bus.
The simplest is those boards using the STorM32 serial driver, which is even a better one than the MAVLink - actually I would say everyone should use the serial driver.

Pedals2Paddles, 2017-11-23:
If the i2c people aren't using the mavlink driver to begin with, why is this even a discussion?  The only driver that needs to change is the mavlink driver, which it sounds like the i2c people wouldn't be using anyway?

olliw42, 2023-02-03:
closing it since really outdated :)

closed 2023-02-03 by olliw42