ArduPilot/ardupilot #7253: Obstacle avoidance in guided mode's velocity control

sharvashish, 2017-11-20:
#### Issue details
Hi all,

I have attached multiple sonars to my Companion Computer, and sending their data to distance_sensor topic.
I have enabled obstacle avoidance parameters correctly, and everything works perfectly in Loiter and AltHold mode.
But in guided mode, I am sending velocity commands to copter, and without obstacle avoidance enabled, the copter is behaving just fine.
Unfortunately when I enable avoidance, the copter more often than not breaches the fence.

I think the issue is related to this: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/control_guided.cpp#L665

In the guided control loop, if curr_vel_des == vel_des, fence check is simply ignored.
After commenting the above snippet, I have tested the code on SITL, and avoidance works 100% of times.
Can someone please explain why such a logic has been built into guided control loop, whereas no such similar logic could be seen in loiter loop?

#### Version
master
#### Platform
[  ] All
[  ] AntennaTracker
[ Y ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad/Hex

#### Hardware type
Pixhawk

OXINARF, 2017-11-22:
Looks like it was there before adding avoidance. I agree we need to always adjust velocity based on avoidance. Instead of removing the check I would change it to only do lines https://github.com/ArduPilot/ardupilot/blob/f01b4836579500989b267284283aa7edb478ca3d/ArduCopter/control_guided.cpp#L669-L684 when the desired velocity is different from current desired.

By the way, we should also call `avoid.adjust_velocity_z`.

Do you want to do a PR fixing this?

sharvashish, 2017-11-22:
Hi @OXINARF ,

Thank you for your suggestion.
Yes. My idea of commenting the line was obviously radical. I just wanted to point out the issue.
I would definitely incorporate your suggestion and do a PR.

For adjust_velocity_z:
yes even this feature is missing. Can you please help me how to add this function over here. In the meantime, I would go through the Loiter loop and find how adjust_velocity_z has been implemented over there.

sharvashish, 2017-11-22:
Hi @OXINARF ,
I think we should use float get_avoidance_adjusted_climbrate(float target_rate) (https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/Copter.h#L704) instead of adjust_velocity_z function.
What say?

OXINARF, 2017-11-22:
@sharvashish I'll comment in your PR (thanks for that!).

rmackay9, 2017-12-07:
I think this is resolved now so closing. txs!

closed 2017-12-07 by rmackay9