ArduPilot/ardupilot #7255: Quadplane: THR_MIN not obeyed in VTOL Modes

DavidIngraham, 2017-11-20:
#### Issue details
When flying as a VTOL the quadplane library will always command 0 throttle unless Q_VFWD_GAIN is in effect. This is an issue for quadplanes that use an internal combustion engine without an ignition enable switch. For these vehicles, the SERVO3_PWM_MIN needs to be set to a value below idle to allow the pilot to kill the engine remotely. When the vehicle is armed, the throttle output servo should not go below THR_MIN to avoid killing the engine.

However, the forward motor throttle calculation in quadplane commands a hard zero in most cases, and does not constrain this to obey THR_MIN. See [here](https://github.com/ArduPilot/ardupilot/blob/126d5ce5d4073f7f40f387e444ea68e5dc31f46c/ArduPlane/quadplane.cpp#L2292)

I belive the proper fix would be to constrain the output to THR_MIN [here](https://github.com/ArduPilot/ardupilot/blob/126d5ce5d4073f7f40f387e444ea68e5dc31f46c/ArduPlane/servos.cpp#L379)

#### Version
_What version was the issue encountered with_
Plane 3.8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk 2.1

magicrub, 2017-12-12:
Seems like a reasonable thing to do. For this particular one I'd like @Tridge to have a quick peek before merging.

WickedShell, 2017-12-12:
@magicrub did you get a peak from @tridge? You seem to have merged the PR 8 minutes after the comment. Given that the issue can be closed can't it?

magicrub, 2017-12-13:
Discussed with Tridge. PR was merged. Closing.

closed 2017-12-13 by magicrub