ArduPilot/ardupilot #7257: Maintain current altitude if thrust not provided in SET_ATTITUDE_TARGET

rcywongaa, 2017-11-21:
#### Issue details
Currently, SET_ATTITUDE_TARGET messages (for GUIDED_NoGPS mode) are only handled if both attitude and thrust is provided.  
Since GUIDED_NoGPS mode functions similarly to ALT_HOLD mode, it would make sense that if thrust is not provided, the vehicle should default to maintain its current altitude while fulfilling (if possible) the attitude targets.

#### Version
3.5

#### Platform
[  ] All
[  ] AntennaTracker
[v] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Pixhawk

Related commit: https://github.com/ArduPilot/ardupilot/pull/2959

rmackay9, 2017-11-21:
@rcywongaa, is this for ROS integration?  I'm keen to improve our support for ROS ..

rcywongaa, 2017-11-21:
@rmackay9 Not in particular (I have not tried using this with ROS).  Just thought I'd put it down, as per our short discussion on gitter (:point_up: [November 20, 2017 7:19 PM](https://gitter.im/ArduPilot/ardupilot?at=5a12ba2671ad3f873608ce7a)

IamPete1, 2021-08-30:
You can send 0 climb rate in the message, so I think that fixes this issue?

https://github.com/ArduPilot/ardupilot/pull/14918

closed 2021-09-11 by IamPete1