ArduPilot/ardupilot #7261: GPS_LEAPSECONDS_MILLIS gives us wrong system time on RPi Navio2 NMEA GPS

lbreckenridge, 2017-11-21:
#### Issue details
_Please describe the problem, or desired feature_
// the number of GPS leap seconds
#define GPS_LEAPSECONDS_MILLIS 18000ULL
We are getting wrong system time set on navio2 on RPi with an NMEA based GPS.

If we change the above to 
#define GPS_LEAPSECONDS_MILLIS 15000ULL
Then the system time will be correct

#### Version
_What version was the issue encountered with_
Latest version

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Boat
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Navio2 on RPi with AIRMAR NMEA2K GPS

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-11-22:
@WickedShell Can you help here? I'm having a bit of a hard time understanding the math in `make_gps_time`, especially:
https://github.com/ArduPilot/ardupilot/blob/f01b4836579500989b267284283aa7edb478ca3d/libraries/AP_GPS/GPS_Backend.cpp#L91-L95

lbreckenridge, 2017-11-22:
> edb478ca3d/libraries/AP_GPS/GPS_Backend.cpp#L91-L95

Hi

I don't understand the math either.
All I know is that in our original fork of the ardupilot code this worked
with the value of 15
And the current value of 18000 / 1000 does not work for us.

Regards
Lloyd

WickedShell, 2019-04-05:
I can't tell what's happening here at all. That function should be replaced entirely with something much cleaner/real comments. In fact it looks to be entirely wrong to me, the first half of that expression will take the year (which will be in the range of [0, 99], subtract off the number of leap seconds (18 now, or 15 before) which would actually wrap around on a uint32_t. We then start multiplying this number out later which makes no sense. As far as I can tell we are also the only people who have that specific constant for convert to time since GPS epoch.

@lbreckenridge when you say it's off is it off by 3 seconds, or off by ~3 days? (based on the code I'm guessing the latter)