ArduPilot/ardupilot #7267: there may be a problem in lean_angles_to_accel().

luweikxy, 2017-11-22:
Hi, guys, I have find a problem in function lean_angles_to_accel(...).
This function lists in the following line:
https://github.com/ArduPilot/ardupilot/blob/c7c2c4ff2c0231e3511fc02820c6526430bed879/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1016

Suppose force produced by the quad is Fb in body frame, then we can kown that Fb = (0, 0, 1/(cos_theta * cos_rho)), then we can get the Fe in the earth frame, i.e.: Fe = Cb->n * Fb, we can get that:
Fe_x = (-mg)*(sin_theta*cos_phi/cos_theta + sin_phi*sin_rho/cos_pho/cos_theta);
Fe_y = (-mg)*(sin_theta*sin_phi/cos_theta - cos_phi*sin_rho/cos_pho/cos_theta);

but in ardupilot code, Fe is as following which is not equal to above.
Fe_x = (-mg)*(sin_theta*cos_phi/cos_theta + sin_phi*sin_rho/cos_pho);
Fe_y = (-mg)*(sin_theta*sin_phi/cos_theta - cos_phi*sin_rho/cos_pho);

So, which is correct?
thanks a lot!

auturgy, 2017-11-22:
the referenced lines of code are basically just a euler rotation and are mathematically correct. 
Where does your fourth angle 'pho' come from?  Why is a 4th angle necessary for a 3 axes transform?

luweikxy, 2017-11-22:
@auturgy 
I am sorry for my carelessness, "pho" is rho and I have corrected this error.
Yes ,the referenced lines of code are basically just a euler rotation, but, there may be a lack of "sin_rho" just as I mentioned above.
Next, here is detailed derivation:
##1
Suppose force produced by the quad is Fb in body frame, then we can easily know that Fb = (0, 0, -mg/(cos_theta * cos_rho)).
##2
 we can get the Fe in the earth frame, i.e.: Fe = Cb->n * Fb
##3
Cb->n =
[xxx              xxx                   sin_theta * cos_phi * cos_rho + sin_phi * sin_rho]
[xxx              xxx                   sin_theta * sin_phi * cos_rho - cos_phi * sin_rho ]
[xxx              xxx                   xxx                                                                       ]
##4
So, Fe = 
[Fe_x]        [xxx    xxx       sin_theta * cos_phi * cos_rho + sin_phi * sin_rho]    [                  0                      ]
[Fe_y]   =   [xxx   xxx       sin_theta * sin_phi * cos_rho - cos_phi * sin_rho ] *  [                  0                      ]
[Fe_z]        [xxx   xxx        xxx                                                                       ]     [-mg/(cos_theta * cos_rho)]
##5
So Fe = 
Fe_x = (-mg) * (sin_theta * cos_phi / cos_theta + sin_phi * sin_rho / **cos_rho** / cos_theta);
Fe_y = (-mg) * (sin_theta * sin_phi / cos_theta - cos_phi * sin_rho / **cos_rho** / cos_theta);
,which is just as what I have said above.

**BUT**, in  lean_angles_to_accel(...), they are:
Fe_x = (-mg) * (sin_theta * cos_phi / cos_theta + sin_phi * sin_rho / cos_theta);
Fe_y = (-mg) * (sin_theta * sin_phi / cos_theta - cos_phi * sin_rho / cos_theta);

rmackay9, 2017-11-22:
@luweikxy, thanks for the report.  I've ping'd @lthall so I hope he can take a look.

lthall, 2017-11-22:
@luweikxy I think you are right. I will double check again in the morning when I am fresh but I suspect I owe you a drink!!

lthall, 2017-11-23:
So this is what it should be. Could you double check what I have done @luweikxy ?

Thank you very much for the catch and taking the time to look!!!

    // rotate our roll, pitch angles into lat/lon frame
    accel_x_cmss = (GRAVITY_MSS * 100) * (-(_ahrs.cos_yaw() * _ahrs.sin_pitch() / MAX(_ahrs.cos_pitch(),0.5f)) 
- _ahrs.sin_yaw() * _ahrs.sin_roll() / MAX(_ahrs.cos_roll()*_ahrs.cos_pitch(),0.5f));

    accel_y_cmss = (GRAVITY_MSS * 100) * (-(_ahrs.sin_yaw() * _ahrs.sin_pitch() / MAX(_ahrs.cos_pitch(),0.5f)) 
+ _ahrs.cos_yaw() * _ahrs.sin_roll() / MAX(_ahrs.cos_roll()*_ahrs.cos_pitch(),0.5f));

luweikxy, 2017-11-23:
@lthall 
Thanks for your attention!
I even feel a little nervous that you an expert ask me to check it. I  am just a ordinary ardupilot user. HaHa. Just for fun.

If we have gotten the Cb->n matrix right, then I think this is right in the aspect of coordinate transformation.

But, I think here may be still a very small problem in the use of MAX(_ahrs.cos_roll()*_ahrs.cos_pitch(),0.5f)).

We know that the force produced by the QUAD in body frame is Fb = (0, 0, -mg/(cos_theta * cos_rho)). however,  (0, 0, -mg/MAX((cos_theta * cos_rho), 0.5f)) is more exactly right for Fb.

So, when we are doing the operation of Fe = Cb->n * Fb, as a denominator, MAX((cos_theta * cos_rho), 0.5f) can not be eliminated when there is a cos_theta or a cos_rho in the numerator.

That is: 
Fe =
[Fe_x]---[xxx-----xxx-----sin_theta * cos_phi * cos_rho + sin_phi * sin_rho]--[-----0-----]
[Fe_y] = [xxx-----xxx-----sin_theta * sin_phi * cos_rho - cos_phi * sin_rho ] * [-----0-----]
[Fe_z]---[xxx-----xxx---------------xxx---------------------------------------]--[-mg/MAX((cos_theta * cos_rho), 0.5f)]
i.e.
Fe_x = (-mg) * (sin_theta * cos_phi * cos_rho + sin_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f);
Fe_y = (-mg) * (sin_theta * sin_phi * cos_rho - cos_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f);

Now I think this may be more truthful.

But, but... given to the new ardupilot code, in the angle boost function: get_throttle_boosted(...) which listed in the line of following:
https://github.com/ArduPilot/ardupilot/blob/023e3b6e30dae577247fa3360cd1ee244cdd9292/libraries/AC_AttitudeControl/AC_AttitudeControl_Multi.cpp#L213

We find that the Fb is  not simply the (0, 0, -mg/MAX((cos_theta * cos_rho), 0.5f))... but more complicated, Fb.z = -mg / MAX((cos_theta * cos_rho) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f), So the final anser of Fe_x and Fe_y is:

Fe_x = (-mg) * (sin_theta * cos_phi * cos_rho + sin_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f);
Fe_y = (-mg) * (sin_theta * sin_phi * cos_rho - cos_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f);

Is the final answer is right? I am not very sure. Could you double check it?

Thanks again for given attention to a very ordinary ardupilot user!

lthall, 2017-11-23:
@luweikxy 
Thanks for looking over this!
You are correct with your statements but I think you misunderstand the purpose of that function. This is intended to generate the same angle request after the Pos Control library is initialized so there is no twitch.

So when these accelerations are used this part of the code will produce the same requested angles:
https://github.com/ArduPilot/ardupilot/blob/c7c2c4ff2c0231e3511fc02820c6526430bed879/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1009

The position controller operates under the assumption that Alt_Hold is keeping the vertical force equal to the weight of the aircraft at all times. It is an approximation I realise.

rmackay9, 2017-11-23:
I've created a PR with Leonard's fix in it. https://github.com/ArduPilot/ardupilot/pull/7273

I'll wait until there's agreement that this is correct before pushing to master.

luweikxy, 2017-11-23:
@lthall 
Many thanks for your explanation!
I am indeed not so familiar with this code yet, so, the following words I said may have mistakes.

Suppose the quad is now in the attitude of euler angle of (80, 0, 0) degree, i.e.: theta angle exceeds 60 degree and is almost vertical.

Then in function lean_angles_to_accel(), we may have three situations to suppose the Fb.

# situation 1
if Fb is supposed as following without any limitations:

Fb = (0, 0, -mg/(cos_theta * cos_rho)).

then Fe is
Fe_x = (-mg) * (sin_theta * cos_phi / cos_theta + sin_phi * sin_rho / cos_rho / cos_theta);
Fe_y = (-mg) * (sin_theta * sin_phi / cos_theta - cos_phi * sin_rho / cos_rho / cos_theta);

, then from the function  [accel_to_lean_angles()](https://github.com/ArduPilot/ardupilot/blob/c7c2c4ff2c0231e3511fc02820c6526430bed879/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1009),
, we will get the _pitch_target is **80**(suppose that there is no lean_angle_max limitation).

# situation 2
if Fb is supposed as following without the limitation of MAX((cos_theta * cos_rho), 0.5f) :

Fb = (0, 0, -mg/MAX((cos_theta * cos_rho), 0.5f)).

then Fe is
Fe_x = (-mg) * (sin_theta * cos_phi * cos_rho + sin_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f);
Fe_y = (-mg) * (sin_theta * sin_phi * cos_rho - cos_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f);

, then from the function  [accel_to_lean_angles()](https://github.com/ArduPilot/ardupilot/blob/c7c2c4ff2c0231e3511fc02820c6526430bed879/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1009),
, we will get the _pitch_target is **60**(suppose that there is no lean_angle_max limitation).

# situation 3
if Fb is supposed as following without the limitations of MAX((cos_theta * cos_rho), 0.5f)  and constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f):

Fb = (0, 0, -mg/MAX((cos_theta * cos_rho), 0.5f) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f)).

then Fe is
Fe_x = (-mg) * (sin_theta * cos_phi * cos_rho + sin_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f);
Fe_y = (-mg) * (sin_theta * sin_phi * cos_rho - cos_phi * sin_rho) / MAX((cos_theta * cos_rho), 0.5f) * constrain_float(2.0f*cos_theta * cos_rho, 0.0f, 1.0f);

, then from the function  [accel_to_lean_angles()](https://github.com/ArduPilot/ardupilot/blob/c7c2c4ff2c0231e3511fc02820c6526430bed879/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1009),
, we will get the _pitch_target is nearly **0**(suppose that there is no lean_angle_max limitation).

==========================================================

So, from the three situations, we can concluded that when the titl angle exceeds 60 degree, if we think Fb has limitation just as situation 3, the _pitch_target and _roll_target will be smaller when the titl angle which exceeds 60 deg is becoming bigger.

And this conclusion just satisfies one of our desire: When the titl angle is exseeding 60 degree, we want the titl angle to get smaller soon.

So, one word, situation 123 are all theoretical OK, but if Fb has limitations, especially limitations as situation 3, the target angle will be smaller when the titl angle is exceeding  60° and is getting near 90°. This is just what we want to get the titl angle smaller soon.

So, I think situation 3 is better and more reasonable. BUT, I am not sure whether this can lead to twitch because the target angle has changed. BUT, in this extreme unstabilized situation, we want the tilt angle get smaller first, not no twitch first.

I am not sure what I said above is right, and I am not sure if I have explained clearly.

OXINARF, 2017-11-24:
#7273 has been merged, closing.

closed 2017-11-24 by OXINARF

luweikxy, 2017-11-25:
@lthall
What do you finally think about it? 
I want to konw the final answer of yours because I am curious about it.
Thanks!

lthall, 2017-12-16:
Ok, so I have finally had time to give this the thought it deserves and I will update the code again to reflect the version that has only one limitation and will not distort the results with large lean angles.

    // rotate our roll, pitch angles into lat/lon frame
    accel_x_cmss = (GRAVITY_MSS * 100) * (-_ahrs.cos_yaw() * _ahrs.sin_pitch() * _ahrs.cos_roll() - _ahrs.sin_yaw() * _ahrs.sin_roll()) / MAX(_ahrs.cos_roll()*_ahrs.cos_pitch(), 0.5f);
    accel_y_cmss = (GRAVITY_MSS * 100) * (-_ahrs.sin_yaw() * _ahrs.sin_pitch() * _ahrs.cos_roll() + _ahrs.cos_yaw() * _ahrs.sin_roll()) / MAX(_ahrs.cos_roll()*_ahrs.cos_pitch(), 0.5f);

To answer your question. 

3 is dependent on there being a large lean angle and angle boost running. If we are in ACRO and at a large lean angle then you are probably holding altitude and the Fbz is probably closer to -mg/(cos_theta * cos_rho). Using the result of 3 will also have the effect of the aircraft going through the unconstrained acceleration part of the Fbz curve on it's way to the low angle causing a large acceleration as it levels. So 3 is not a good option in general.

1 and 2 are reasonable and both will result in a good starting point. 2 will result in a twitch as you say but due to the acceleration limiting in most navigation modes 1 will to. In any case we have to limit the angle to something to prevent the cos terms going to zero so we need some version of 2.

For now I will leave it using the 2 version until we update the output of the navigation to a full 3D acceleration controller.

Thanks again and sorry for the delay!