ArduPilot/ardupilot #7279: Plane: Do not initiate RTL or automatically start motor when airplane is not flying

bchristal, 2017-11-24:
#### Issue details
Failsafe events cause airplane to enter RTL when not flying. This is dangerous. A typical scenario is an airplane is armed without powering the RC TX. Or, the RC TX is turned off while an armed airplane is on the ground/workbench. Loss of RC TX causes RTL which "unexpectedly" starts motor.

I can only think of two scenarios where the motor should start on an airplane that is *not flying*.
-Auto Takeoff
-RC TX throttle moves up

*Not flying* airplane motors *shall not auto start motor* when entering flight mode RTL, FBWB, other guided modes?

#### Version
3.8. But this behavior goes way back.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Flying wing, standard wing

#### Hardware type
Pixhawk, any RC Receiver

#### Logs
none provided. easy to duplicate

auturgy, 2017-11-28:
https://discuss.ardupilot.org/t/disable-rc-throttle-failsafe-when-airplane-is-on-the-ground/23393/
Logs in the thread.

pompecukor, 2017-12-03:
I think this would be a welcome feature. That is never to actually engage motors if Plane:
Is armed but is NOT flying.

IT might be tricky as indeed one would want to test FS and RTL on ground. So it is not like disabling RTL when is_flying = 0. Only not to allow motors to spin even if armed.

I set up regularly and have had a few close calls, so I feel it is just a question of time.

bchristal, 2018-02-07:
I just thought I'd bring this up again. I have found a common situation that also causes full throttle starts on the ground. Taranis X7 and L9R and X8R (very common) will lose RC signal if the transmitter and receiver are closer than 1.5 feet. This results in full throttle motor start. Again, incredibly dangerous.

peterbarker, 2021-07-30:
I've replicated the scenario in the linked log file in SITL on master - not exactly, but near enough.

Throttle is becoming unsuppressed as soon as the EKF becomes active:
 
[image]

One of the reasons we unsuppress throttle is because we believe we are >10m, above home....
 
Vehicle thinks it is >10m above as soon as EKF is active:
[image]

DCM would still be using the Baro for height:

[image]

So what this comes to is the code is working as it is currently designed (and was in Feb 2018....).  The vehicle is >10m above home, so unsupresses the throttle.  The fact that it is bad sensor data causing the problem means that the current mechanisms may be insufficient.

Perhaps we should add `is_flying()` to the mix here - we're already deep into heuristic-land with this check.