ArduPilot/ardupilot #7280: Booting with BRD_SAFETYENABLE=0

kd0aij, 2017-11-25:
#### Issue details
undesirable servo movements at bootup

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quadplane_tilttrivec (E-Flite Convergence)

#### Hardware type
pixracer

#### Logs

log of servo outputs from Pixracer with BRD_SAFETYENABLE=0
RCOU.Cn are scaled by .01 (scaled range [100,200])
elevons are channels 2,4, starting PWM values are near 1900
vehicle is stationary until about 38 seconds
elevons take about 35 seconds to settle down

[image]

kd0aij, 2017-11-27:
another test with the same hardware and slightly different results: 

- USB power only
- SERVO1_FUNCTION  76.000000
- SERVO2_FUNCTION  78.000000
- SERVO3_FUNCTION  75.000000
- SERVO4_FUNCTION  77.000000

[image]

WickedShell, 2018-03-28:
@kd0aij any chance that you had AFS enabled? (this can move surfaces during boot, although why it would move a tilt rotor motor is an interesting question)

Naterater, 2019-01-05:
What mode are you in on boot? If you boot to a stabilization mode while the gyros are not consistent, you may have "stabilizing" behavior due to bad AHRS. Maybe consider booting in Manual Mode?

kd0aij, 2019-01-05:
@Naterater Is it possible to boot into manual mode without setting up manual mode on the RC TX?
I suppose I could boot with the TX off...

Naterater, 2019-01-17:
Yes it is possible. If so, that would explain this issue and we can close. I don't think anybody else has brought this up, and it's likely because of the EKF settling down after boot. I notice my artificial horizon moves a lot on boot.

kd0aij, 2019-01-17:
So what is your power-up sequence? TX-on, then vehicle power up?
At what point will it switch from boot mode to the one commanded by RC?
It's usually unwise to power up a vehicle with the TX off.

OXINARF, 2019-07-12:
Is this still relevant with ChibiOS?

WickedShell, 2019-07-12:
I've seen servos get a pulse that looked like a normal PWM while booting exactly once on ChibiOS. Safety was enabled, but it's a different thing to what's described here as this is all before logging starts to be able to log what's happening (and it was only one channel that got a valid shaped pulse).

kd0aij, 2019-07-12:
Behavior hasn't changed; elevons and tilt servos are active immediately after bootup.
Workaround for tiltrotors is to set INITIAL_MODE to a VTOL mode to avoid having the motors tilt forward. 
Having the elevons wander around isn't a big deal and does have the advantage of showing the EKF starting up.