ArduPilot/ardupilot #7304: All: add loop logging

sashan13, 2017-11-27:
#### Issue details
So, the issue is that when SD card is full of logs on Pixhawk the logging stops. This happens fairly quickly when we are logging many parameters at once, like IMU data.

This is crucial because it may result in being unable to analyze crash, or in a failed mapping mission. In latter case, it means lost time and money. 

I propose to make it delete the oldest log when it detects that there are, e.g. less than 1mb of free space on SD card.  Losing oldest log is much better than losing newest one. Alternatively, there may be a parameter that selects between those behaviors. 

#### Version
AC3.5.3

#### Platform
[x] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Copter - but probably all.

#### Hardware type
Pixhack

#### Logs
No logs were recorded

WickedShell, 2017-11-27:
The autopilot already attempts to delete old log files when there is less then 10% diskspace on boot. This doesn't happen in flight due to not wanting to have to many file descriptors open, and any files other then log files won't be deleted. Are you not seeing this work for some reason, or are you logging mode data then 10% of your disk capacity?

sashan13, 2017-11-27:
I don't see it work. Consistently, with all AP versions starting with 3.4.
I am using 8gb SD cards that have nothing but logs (and a bit of terrain
data) on them.

peterbarker, 2017-11-28:
On Mon, 27 Nov 2017, sashan13 wrote:

> I propose to make it delete the oldest log when it detects that there are,
> e.g. less than 1mb of free space on SD card. Losing oldest log is much

We don't delete stuff in flight.  The filesystem code isn't something to 
mess with lightly.

https://github.com/ardupilot/ardupilot/blob/master/libraries/DataFlash/DataFlash_File.cpp#L350

You should see messages at boot indicating it is deleting files for space 
requirements:

https://github.com/ardupilot/ardupilot/blob/master/libraries/DataFlash/DataFlash_File.cpp#L383

OXINARF, 2017-12-01:
@peterbarker Just so I understand if we should close this or leave it as an enhancement: are you saying that we should never delete logs in flight or are you saying that the current NuttX filesystem implementation isn't safe for us to do that? I'm thinking in the ChibiOS port, if it will be safe to do there.

peterbarker, 2017-12-01:
Only a problem on NuttX at the moment.  I don't think it would be a problem on Linux.

Having said that, I'd want to move responsibility for doing file operations into the IO thread as part of this.  Or at least do this bit in the IO thread.

IamPete1, 2021-08-08:
We now have a param to specify the amount of space you want free on the SD card, AP will remove old logs until there is that amount of space.

closed 2021-08-08 by IamPete1