ArduPilot/ardupilot #7324: Rover: Simultaneous rate and speed setting is broken

pavloblindnology, 2017-11-29:
Due to https://github.com/ArduPilot/ardupilot/commit/3863542194e77e9ce690aa8bc86577fce984d54f simultaneous setting of rate and speed is broken in guided mode for `MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED` and `MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT`.
Propose to add smth like this in `// set guided mode targets` sections

> // set guided mode targets
> ...
> } else if (pos_ignore && !vel_ignore && acc_ignore && yaw_ignore && !yaw_rate_ignore) {
>     // consume speed and turn rate
>     rover.mode_guided.set_desired_turn_rate_and_speed(target_turn_rate_cds, packet.vy);
> }
> 
https://github.com/ArduPilot/ardupilot/blob/68800a52f5c73fc0c0b4b94357a3198ba16bf60d/APMrover2/GCS_Mavlink.cpp#L1141

khancyr, 2017-11-29:
Hello,

Yes, are aware of that. I will push the correction tonight !

pavloblindnology, 2017-11-29:
Thank you @khancyr

OXINARF, 2017-12-02:
Duplicate of #7217

closed 2017-12-02 by OXINARF