ArduPilot/ardupilot #7327: Copter: Is there any problem in function euler_accel_limit()?

luweikxy, 2017-11-30:
ArduPilot/ardupilot/blob/ed0063d10dd690529fadeabdc35ca5cb32815139/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L530

the function euler_accel_limit(...) is to translate the angle accel from body frame to NED frame, but I cann't undersdand this function.

Whereas I think the right way is like the euler kinematical equation, just use angle accel instead of angle rate.

the euler kinematical equation is as following:
[image]

lthall, 2017-11-30:
Hi @luweikxy,

Thanks again for going over this so thoroughly!

So the problem here is we need to calculate the acceleration limits that will not let us increase past any one individual axis acceleration limit. I think you are calculating the maximum acceleration limits possible, not the minimum that we need.

So for example, if we are banked over by 45 degrees in roll. If we ask apply full yaw then the acceleration will be shared by both the body pitch and body yaw. However, if we apply full yaw and push the pitch stick forward then that euler yaw acceleration is applied only to the body frame yaw and is increased in magnitude because the aircraft is banked by 45 degrees. So we need to limit the yaw acceleration based on this limit rather than the vector addition of both body pitch and body yaw.

I realise this is poorly worded but does it make sense?

Thanks again!

lthall, 2017-12-01:
I have been thinking about this. Perhaps what we need to do here is look at the maximum possible acceleration in the direction of that acceleration rather than the maximum safe acceleration in each axis with no knowledge of what is happening in any other axis.

The function would need a little more information though.

IamPete1, 2021-09-12:
@lthall Is there any action we should take due to this? It sounds like the there is not a problem so we should close, but maybe there is a enhancement that we could open a new issue for?

closed 2021-09-12 by lthall

luweiagi, 2023-03-07:
final solution: [Copter: Is there any problem in function euler_accel_limit() ? version-2](https://github.com/ArduPilot/ardupilot/issues/23117)

lixiaowei123, 2023-07-28:
@lthall He wanted to ask how the formula was derived, but you didn't answer his question

lthall, 2023-07-28:
@lixiaowei123 I don't agree.