ArduPilot/ardupilot #7330: Music tones after firmware upload

aryklein, 2017-11-30:
After finish the firmware upload (Arduplane 3.8.3) using Mission Planner in a Pixhawk board, it doesn't play the music it used to play with previous versions (for example with Arduplane 3.8.0). It only plays the normal boot sound. Anyway if I go into messages I see it's running Arduplane 3.8.3 as expected.
Does the new version remove the music after FW upload?

OXINARF, 2017-12-05:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

jmachuca77, 2017-12-05:
I have experienced this too. After firmware upload the pixhawk used to play a confirmation that the firmware upload is complete. 

http://download.ardupilot.org/downloads/wiki/pixhawk_sound_files/PX4_CompletedIOBoardFirmwareUpload.wav

This no longer plays reliably, or its played but its cut in the middle by the regular power on tones. I think this is a bug and not a user support issue.

OXINARF, 2017-12-05:
As the filename says that is the tone it plays after the IO is updated. If there is no new IO firmware then there is nothing to update and the tone isn't played.

Being cut in the middle by other tones just seems possible for the final tone (https://jschall.github.io/#MSPAA). We could add a delay so that the tone is never cut, but it doesn't seem worth it to me to delay initialization to play that tone.

closed 2017-12-05 by OXINARF

rmackay9, 2017-12-05:
let's not fight over opening and closing this issue.  If someone on the dev team wants to leave it open, let's leave it open.