ArduPilot/ardupilot #7331: ::create methods - not calling new() and thus uninitialised memory used for objects

peterbarker, 2017-12-01:
#### Issue details
I'm getting segmentation faults in AP_SBusOut.  In the constructor, `sbus1_uart` has the value 0x5.  If `new()` had been called that would be 0x0.

AFAICS the standards seem to allow calls to new() to be optimised out.  Since `AP_SBusOut::create()` allocates the object on the stack it is likely that's happening (see disassembly below).  Certainly in C++14 it has been clarified behaviour that `new()` can be omitted.

@lucasdemarchi - were we relying on our compilers to NOT omit the call to `new()` for this `::create()` stuff to work?  If so, we might need a new plan....

#### Version
master

#### Platform
[ X ] All

#### Airframe type
All

#### Hardware type
SITL

#### Logs

This was produced on my laptop with debug symbols but with `-O3` (by hacking `ardupilotwaf.py`):
```
Dump of assembler code for function SRV_Channels::SRV_Channels():
   0x00005555555ca1a0 <+0>:     push   %r14
   0x00005555555ca1a2 <+2>:     push   %r13
   0x00005555555ca1a4 <+4>:     lea    0x8(%rdi),%r13
   0x00005555555ca1a8 <+8>:     push   %r12
   0x00005555555ca1aa <+10>:    push   %rbp
   0x00005555555ca1ab <+11>:    mov    %rdi,%r12
   0x00005555555ca1ae <+14>:    push   %rbx
   0x00005555555ca1af <+15>:    mov    %r13,%rdi
   0x00005555555ca1b2 <+18>:    lea    0x38(%r12),%r14
   0x00005555555ca1b7 <+23>:    lea    0x118(%r12),%rbp
   0x00005555555ca1bf <+31>:    sub    $0x20,%rsp
   0x00005555555ca1c3 <+35>:    mov    %r14,%rbx
   0x00005555555ca1c6 <+38>:    mov    %fs:0x28,%rax
   0x00005555555ca1cf <+47>:    mov    %rax,0x18(%rsp)
   0x00005555555ca1d4 <+52>:    xor    %eax,%eax
   0x00005555555ca1d6 <+54>:    callq  0x5555555dd870 <AP_Volz_Protocol::AP_Volz_Protocol()>
=> 0x00005555555ca1db <+59>:    mov    %rsp,%rdi
   0x00005555555ca1de <+62>:    callq  0x5555555ddd80 <AP_SBusOut::AP_SBusOut()>
   0x00005555555ca1e3 <+67>:    mov    (%rsp),%rax
   0x00005555555ca1e7 <+71>:    mov    %rax,0x28(%r12)
   0x00005555555ca1ec <+76>:    movzwl 0x8(%rsp),%eax
   0x00005555555ca1f1 <+81>:    mov    %ax,0x30(%r12)
   0x00005555555ca1f7 <+87>:    movzwl 0xa(%rsp),%eax
   0x00005555555ca1fc <+92>:    mov    %ax,0x32(%r12)
   0x00005555555ca202 <+98>:    movzbl 0xc(%rsp),%eax
   0x00005555555ca207 <+103>:   mov    %al,0x34(%r12)
   0x00005555555ca20c <+108>:   nopl   0x0(%rax)
   0x00005555555ca210 <+112>:   mov    %rbx,%rdi
   0x00005555555ca213 <+115>:   add    $0xe,%rbx
   0x00005555555ca217 <+119>:   callq  0x555555646550 <SRV_Channel::SRV_Channel()>
   0x00005555555ca21c <+124>:   cmp    %rbp,%rbx
   0x00005555555ca21f <+127>:   jne    0x5555555ca210 <SRV_Channels::SRV_Channels()+112>
   0x00005555555ca221 <+129>:   lea    0x2cff98(%rip),%rsi        # 0x55555589a1c0 <_ZN12SRV_Channels8var_infoE>
```

lucasdemarchi, 2017-12-01:
::create() will just hide the call to new() and nothing else. It should have the same behavior as before.

Most times I've got crash on constructors it's actually due to calling uninitialized global static objects rather than anything else.... And the fix is usually to move that stuff to the init() method. I didn't look deep in the issue, but isn't this the case? Can you post a backtrace and valgrind log?

peterbarker, 2017-12-01:
On Fri, 1 Dec 2017, Lucas De Marchi wrote:

> ::create() will just hide the call to new() and nothing else. It should have
> the same behavior as before.

I believe what is happening is the compiler is simply deciding NOT to 
call new() because it is allocating the object on the stack.  It is 
allowed to do that according to the spec.

The disassembly doesn't seem to have anything in it which would zero 
chunks of memory.

> Most times I've got crash on constructors it's actually due to calling
> uninitialized global static objects rather than anything else.... And the
> fix is usually to move that stuff to the init() method. I didn't look deep

We absolutely rely on member variables to be zero-initialised (by 
the new() override) at the moment, or by the implicit bss initialisation.

> in the issue, but isn't this the case? Can you post a backtrace and valgrind
> log?

I'll need to put together a VM for that.  Gimme a while....

lucasdemarchi, 2017-12-04:
> I believe what is happening is the compiler is simply deciding NOT to
> call new() because it is allocating the object on the stack.  It is
> allowed to do that according to the spec.

That is a *static* object. So you are relying on the object being placed on the bss section. There's no call to new() involved.

peterbarker, 2017-12-05:
On Mon, 4 Dec 2017, Lucas De Marchi wrote:

> I believe what is happening is the compiler is simply deciding NOT to
>       call new() because it is allocating the object on the stack. It is
>       allowed to do that according to the spec.
>
> That [...]

In what way is it static?  Notionally it is being created on the stack and 
returned AFAICS.  But the copy/move is being elided, because apparently 
you can do that (C++ section 12.8 - not that I understand a tiny fraction 
of it).

IOW:

     static AP_SBusOut create() {
         return AP_SBusOut{};
     }

should be equivalent to:

     static AP_SBusOut create() {
         auto foo = AP_SBusOut{};
         return foo;
     }

And that foo is on the stack, meaning that new() isn't called and the 
copy-ellision means that the deleted copy constructor isn't an issue.

lucasdemarchi, 2017-12-05:
Yes, those 2 are equivalent... you are just forgetting you need to assign the result to something, i.e. whatever called the method create().

```c++
static AP_SBusOut sbus = AP_SBusOut::create();
```

Or if the sbus object is inside another **static** object. In this case it's inside SRV_Channels, which is inside the vehicle (or ParametersG2) and that one should be on bss.

lucasdemarchi, 2017-12-05:
```c++
#include <stdio.h>

class Foo {
public:
    ~Foo() { printf("destroy\n"); }
    static Foo create() { return Foo{}; }

    // This should be zero when inside an static object.
    int test_zero;

private:
    Foo() { printf("create %d\n", Foo::n); Foo::n++; }

    static int n;
};

int Foo::n;

class Vehicle {
public:
    Vehicle() {}
    Foo foo = Foo::create();
};

static Vehicle v;

int main()
{
    printf(" test_zero on Foo inside static object = %d\n", v.foo.test_zero);

    // valgrind should complain loudly below
    Foo foo2 = Foo::create();
    printf(" test_zero on an object in the stack = %d\n", foo2.test_zero);

    static Foo foo3 = Foo::create();
    printf(" test_zero on static object = %d\n", foo3.test_zero);

    return 0;
}
```

This is the closest to what we have in the codebase. Please give this a try.

peterbarker, 2017-12-09:
On Tue, 5 Dec 2017, Lucas De Marchi wrote:

> #include <stdio.h>
> 
> class Foo {
.
.
> This is the closest to what we have in the codebase. Please give this a
> try.

That's clean according to Valgrind.

Might need to delete the various copy constructors and the like - as we 
do in the AP code?

I'll try to find time to play with your example.

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

tridge, 2017-12-11:
this is the valgrind report
==1908== Conditional jump or move depends on uninitialised value(s)
==1908==    at 0x4BAC88: AP_SBusOut::update() (AP_SBusOut.cpp:84)
==1908==    by 0x4A3BB6: SRV_Channels::push() (SRV_Channels.cpp:212)
==1908==    by 0x42C02C: Plane::servos_output() (servos.cpp:683)
==1908==    by 0x41CAE1: Plane::failsafe_check() (failsafe.cpp:91)
==1908==    by 0x42CA04: failsafe_check_static() (system.cpp:17)
==1908==    by 0x4E85DC: HALSITL::Scheduler::_run_timer_procs(bool) (Scheduler.cpp:178)
==1908==    by 0x4E656E: HALSITL::Scheduler::timer_event() (Scheduler.h:47)
==1908==    by 0x4E6DF1: HALSITL::SITL_State::_fdm_input_step() (SITL_State.cpp:182)
==1908==    by 0x4E6E3B: HALSITL::SITL_State::wait_clock(unsigned long) (SITL_State.cpp:190)
==1908==    by 0x4E819B: HALSITL::Scheduler::delay_microseconds(unsigned short) (Scheduler.cpp:45)
==1908==    by 0x4E81D6: HALSITL::Scheduler::delay(unsigned short) (Scheduler.cpp:52)
==1908==    by 0x42E103: Plane::startup_INS_ground() (system.cpp:559)

tridge, 2017-12-11:
Peter and I have looked into this, and as far as we can see this is a fundamental problem with the create() approach.
The AP_SBusOut::create() method creates a AP_SBusOut object on the stack:
    static AP_SBusOut create() {
        return AP_SBusOut{};
    }
that is saying "create a AP_SBusOut object on the stack and then return a copy of it
When it creates it on the stack it does not run either new or use the bss.  So if the address on the stack corresponding to the sbus1_uart ptr is non-null then this copies an invalid pointer into copter.g2.servo_channels.sbus.sbus1_uart
@lucasdemarchi I'd like to discuss this with you urgently, as I believe that right now master is vulnerable to uninitialized variables in a way that we were not vulnerable before we added create().

peterbarker, 2017-12-11:
```
Starting program: /home/pbarker/rc/ardupilot/build/sitl-debug/bin/arducopter --model quad -C
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Hardware watchpoint 2: copter.g2.servo_channels.sbus.sbus1_uart

Old value = (AP_HAL::UARTDriver *) 0x0
New value = (AP_HAL::UARTDriver *) 0x7fffffffdb90
0x00005555555f15dc in SRV_Channels::SRV_Channels (
```

```
Dump of assembler code for function SRV_Channels::SRV_Channels():
   0x00005555555f1590 <+0>:	push   %r14
   0x00005555555f1592 <+2>:	push   %r13
   0x00005555555f1594 <+4>:	lea    0x8(%rdi),%r13
   0x00005555555f1598 <+8>:	push   %r12
   0x00005555555f159a <+10>:	push   %rbp
   0x00005555555f159b <+11>:	mov    %rdi,%r12
   0x00005555555f159e <+14>:	push   %rbx
   0x00005555555f159f <+15>:	mov    %r13,%rdi
   0x00005555555f15a2 <+18>:	lea    0x38(%r12),%r14
   0x00005555555f15a7 <+23>:	lea    0x118(%r12),%rbp
   0x00005555555f15af <+31>:	sub    $0x20,%rsp
   0x00005555555f15b3 <+35>:	mov    %r14,%rbx
   0x00005555555f15b6 <+38>:	mov    %fs:0x28,%rax
   0x00005555555f15bf <+47>:	mov    %rax,0x18(%rsp)
   0x00005555555f15c4 <+52>:	xor    %eax,%eax
   0x00005555555f15c6 <+54>:	callq  0x555555604fe0 <AP_Volz_Protocol::AP_Volz_Protocol()>
   0x00005555555f15cb <+59>:	mov    %rsp,%rdi
   0x00005555555f15ce <+62>:	callq  0x5555556054f0 <AP_SBusOut::AP_SBusOut()>
   0x00005555555f15d3 <+67>:	mov    (%rsp),%rax
   0x00005555555f15d7 <+71>:	mov    %rax,0x28(%r12)
=> 0x00005555555f15dc <+76>:	movzwl 0x8(%rsp),%eax
   0x00005555555f15e1 <+81>:	mov    %ax,0x30(%r12)
   0x00005555555f15e7 <+87>:	movzwl 0xa(%rsp),%eax
   0x00005555555f15ec <+92>:	mov    %ax,0x32(%r12)
   0x00005555555f15f2 <+98>:	movzbl 0xc(%rsp),%eax
   0x00005555555f15f7 <+103>:	mov    %al,0x34(%r12)
   0x00005555555f15fc <+108>:	nopl   0x0(%rax)
   0x00005555555f1600 <+112>:	mov    %rbx,%rdi
   0x00005555555f1603 <+115>:	add    $0xe,%rbx
   0x00005555555f1607 <+119>:	callq  0x55555568c7e0 <SRV_Channel::SRV_Channel()>
   0x00005555555f160c <+124>:	cmp    %rbp,%rbx
   0x00005555555f160f <+127>:	jne    0x5555555f1600 <SRV_Channels::SRV_Channels()+112>
   0x00005555555f1611 <+129>:	lea    0x2fdec8(%rip),%rsi        # 0x5555558ef4e0 <_ZN12SRV_Channels8var_infoE>
   0x00005555555f1618 <+136>:	mov    %r12,%rdi
   0x00005555555f161b <+139>:	mov    %r12,0x306d36(%rip)        # 0x5555558f8358 <_ZN12SRV_Channels8instanceE>
   0x00005555555f1622 <+146>:	mov    %r14,0x306d37(%rip)        # 0x5555558f8360 <_ZN12SRV_Channels8channelsE>
   0x00005555555f1629 <+153>:	add    $0x28,%r12
```

peterbarker, 2017-12-11:
```
(gdb) bt
#0  0x00005555555f15dc in SRV_Channels::SRV_Channels (
    this=0x5555558f2ad0 <copter+1136>)
    at ../../libraries/SRV_Channel/SRV_Channels.cpp:131
#1  0x0000555555578239 in ParametersG2::ParametersG2 (
    this=0x5555558f2790 <copter+304>) at ../../ArduCopter/Parameters.cpp:1045
#2  0x0000555555571777 in Copter::Copter (this=0x5555558f2660 <copter>)
    at ../../ArduCopter/Copter.cpp:60
#3  0x000055555556c70c in __static_initialization_and_destruction_0 (
    __initialize_p=1, __priority=65535) at ../../ArduCopter/Copter.cpp:69
#4  _GLOBAL__sub_I_hal () at ../../ArduCopter/Copter.cpp:69
#5  0x00005555556a624d in __libc_csu_init ()
#6  0x00007ffff6f6a380 in __libc_start_main (
    main=0x55555556c6c0 <main(int, char* const*)>, argc=4, 
    argv=0x7fffffffdee8, init=0x5555556a6200 <__libc_csu_init>, 
    fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7fffffffded8)
    at ../csu/libc-start.c:247
#7  0x000055555556e17a in _start ()
(gdb) 
```

lucasdemarchi, 2017-12-12:
> When it creates it on the stack it does not run either new or use the bss.

It should be called for objects that are on bss or that are dynamically allocated since it will just initialize the fields.

I see where this can possibly go wrong with the compiler if there is in fact an object instantiated on the stack with garbage and this is copied over the original place. However this should not happen as it should be initializing the object to which this is assigned to rather than copying the entire object.  From the disassembly it looks like the wrong thing is happening and
```asm
   0x00005555555f15d3 <+67>:	mov    (%rsp),%rax
```
is assigning the pointer on the stack to the final object :-/.

I'm wondering why this is different for this object since the others are/were working. This is the case, for example, for AP_Volz_Protocol on this same function. It's possible to disable this single user and see if the the problem goes away or if there's something else: 
https://gist.github.com/4f9c6e00aa8e6391764e936af5436772

If it goes away then we need to re-evaluate the ::create method.

lucasdemarchi, 2017-12-12:
doing so I do get the correct behavior:

```asm
  483916:       48 83 ec 08             sub    $0x8,%rsp
  48391a:       e8 31 43 01 00          callq  497c50 <AP_Volz_Protocol::AP_Volz_Protocol()>
  48391f:       4c 89 fb                mov    %r15,%rbx
  483922:       4c 89 ef                mov    %r13,%rdi
  483925:       e8 56 48 01 00          callq  498180 <AP_SBusOut::AP_SBusOut()>
  48392a:       66 0f 1f 44 00 00       nopw   0x0(%rax,%rax,1)
```

/me scratching his head

tridge, 2017-12-12:
Hi Lucas,
I think it's only working for everything else by luck. What create() does for objects that are in bss is to change them from getting values that come from bss to getting values that come from the stack.
The only thing that puzzles me is why we aren't getting valgrind warnings for all the other use cases.
Cheers, Tridge

lucasdemarchi, 2017-12-12:
Not exactly luck, I'm never so lucky after all :) It should be doing the copy elision, initializing the object in place. In this case it would be initialized.

tridge, 2017-12-12:
but the variables are not explicitly initialised, so what would be initalising them? Are you saying we're relying on the syntax of object{} to zero all variables that aren't explicitly set?

lucasdemarchi, 2017-12-12:
no, that it should be initializing the object in place (i.e. either an object from bss or that has been zero-initialized by us), not copying an entire object over to the final location. This is what it is doing in other places.

OXINARF, 2017-12-12:
@lucasdemarchi I think you meant move elision, as copy operation is forbidden for all classes with the create method.

@peterbarker @tridge Does this only happen with a version compiled with debug info or also a "release" version? The reason I ask is that, apparently, for debug builds the compilers usually don't do elisions - still strange why it would only happen with AP_SbusOut.

tridge, 2017-12-12:
so what makes it invalid for the compiler to create the object on the stack and return that object?
ahh, its the object size! If I add a few more bytes of unused variables to the AP_SBusOut object then the valgrind error goes away.
So looks like gcc optimises differently for small objects

OXINARF, 2017-12-12:
Wow, great catch!

I should add that this type of copy/move elision is actually mandatory after C++17. But we need to decide what to do now.

tridge, 2017-12-12:
this seems to imply that the create() approach in our code is unsafe. Should we start working on removing it now? Or can someone suggest how to make it safe?

lucasdemarchi, 2017-12-12:
Humn.... good catch. One thing we could do is to use the move constructor explicitly rather than doing the assignment...

OXINARF, 2017-12-12:
@lucasdemarchi How would that solve the problem? I think the compiler would still (or at least be allowed to) be constructing the temporary in stack and then move it (with uninitialized values).

@tridge I agree that this is unsafe and we need to fix it. Removing the create though is changing a big pile of lines again, so I think we should try to come up with a different solution first.

tridge, 2017-12-12:
> Removing the create though is changing a big pile of lines again, so I think we should try to come up with a different solution first.

I'm certainly open to suggestions, but I think this has also shown that getting this right is more complex than we originally thought. I'm starting to wonder if the reduction in static analyzer noise is worth the complexity.

lucasdemarchi, 2017-12-12:
> I'm certainly open to suggestions, but I think this has also shown that getting this right is more complex than we originally thought. I'm starting to wonder if the reduction in static analyzer noise is worth the complexity.

If you look in the PR where this got merged it actually caught real bugs that could be as bad as this one. I'll try to come up with a solution shortly

lucasdemarchi, 2017-12-12:
I think the best compromise here is to remove the ::create() methods, but keep the deleted constructors (those were actually found problems in the codebase). Something like this:

```diff
diff --git a/libraries/AP_SBusOut/AP_SBusOut.h b/libraries/AP_SBusOut/AP_SBusOut.h
index 9e0de771b1..a30f8eaf86 100644
--- a/libraries/AP_SBusOut/AP_SBusOut.h
+++ b/libraries/AP_SBusOut/AP_SBusOut.h
@@ -15,9 +15,7 @@ class AP_SBusOut {
 public:
     static const struct AP_Param::GroupInfo var_info[];
 
-    static AP_SBusOut create() {
-        return AP_SBusOut{};
-    }
+    AP_SBusOut();
 
     constexpr AP_SBusOut(AP_SBusOut &&other) = default;
 
@@ -28,7 +26,6 @@ public:
     void update();
 
 private:
-    AP_SBusOut();
     AP_HAL::UARTDriver *sbus1_uart;
 
     void init(void);
diff --git a/libraries/SRV_Channel/SRV_Channel.h b/libraries/SRV_Channel/SRV_Channel.h
index bc57b901d5..b9b13afec5 100644
--- a/libraries/SRV_Channel/SRV_Channel.h
+++ b/libraries/SRV_Channel/SRV_Channel.h
@@ -442,7 +442,7 @@ private:
     static AP_Volz_Protocol *volz_ptr;
 
     // support for SBUS protocol
-    AP_SBusOut sbus = AP_SBusOut::create();
+    AP_SBusOut sbus{};
     static AP_SBusOut *sbus_ptr;
 
     SRV_Channel obj_channels[NUM_SERVO_CHANNELS];
```

The generated assembly is here:

```asm
  48391f:       4c 89 fb                mov    %r15,%rbx
  483922:       4c 89 ef                mov    %r13,%rdi
  483925:       e8 56 48 01 00          callq  498180 <AP_SBusOut::AP_SBusOut()>
  48392a:       66 0f 1f 44 00 00       nopw   0x0(%rax,%rax,1)
```

Note above that we are not copying the values from stack to the final place anymore. I can work on this and provide the patches if it looks ok.

tridge, 2017-12-12:
thanks @lucasdemarchi, I'm working on a PR now to implement this

tridge, 2017-12-13:
PR is here:
https://github.com/ArduPilot/ardupilot/pull/7401

OXINARF, 2017-12-18:
Fixed by #7401.

closed 2017-12-18 by OXINARF