ArduPilot/ardupilot #7333: Rally points outside geofence force fence breach and crash

Pedals2Paddles, 2017-12-01:
#### Issue details
If you have rally points set, and they happen to be outside the geofence, this can force a fence breach and undesired landing.

_Scenario:_  You're flying around inside the geofence. A rally point outside the geofence is closer to the home when RTL is activated.  The copter will proceed towards the rally point and breach the geofence.  The geofence failsafe is RTL, which again makes it head for the rally point that is outside the geofence.  The copter then blasts right past the second stage of the geofence, which switches the copter to land mode.  Now it blindly lands on whatever happens to be below it.

_Suggestions:_ Verify rally points are not outside the fence in the arming check?  

#### Version
Copter 3.5, but I would imagine this could happen with any vehicle

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
N/A

rmackay9, 2017-12-01:
Thanks for the report.  I'm sure we have talked about this possibility but I think we didn't realise that it would switch to LAND mode.

Pedals2Paddles, 2017-12-01:
I agree now that I think about it some more.  Allow arming and operationally ignore the ones outside the fence. Would it be a heavy lift to detect this in the pre-arm checks and at least send a caution message to the GCS?