ArduPilot/ardupilot #7341: Support HIGH_LATENCY data stream

envirover, 2017-12-02:
#### Issue details
Currently HIGH_LATENCY MAVLink message is not supported by ArduPilot. HIGH_LATENCY message is used by global satellite telemetry SPL (see https://github.com/envirover/SPLRadioRoom). The current implementation or SPL RadioRoom composes HIGH_LATENCY messages form several different messages by requesting multiple data streams. This is not a very efficient and reliable solution.

#### Version
All

#### Platform
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All

#### Hardware type
All

DavidIngraham, 2017-12-02:
I would like to +1 this. We will soon have vehicles using iridium, and this message would simplify things greatly.

It will also require support on the GCS side.

dagar, 2017-12-04:
I'll likely be adding QGC support for the basic HIGH_LATENCY message in the near future. https://github.com/mavlink/qgroundcontrol/pull/4443

The harder part will be making it aware of the special connection type and acting accordingly.

DavidIngraham, 2017-12-08:
How would heartbeats be handled when sending this message? Would we only send the heartbeat to initiate the connection, then treat the HIGH_LATENCY message as the heartbeat from then on? The GCS will need a heartbeat in the beginning to establish the autopilot family, vehicle type,  and protocol version.

dagar, 2017-12-09:
Partially TBD, but what I've discussed with Don is making HIGH_LATENCY (or a slightly changed 2nd iteration) a substitute for HEARTBEAT.

dagar, 2017-12-22:
Working on a second iteration of HIGH_LATENCY to fix a few shortcomings. https://github.com/mavlink/mavlink/pull/814

amilcarlucas, 2020-12-09:
HIGH_LATENCY is now deprecated, please implement HIGH_LATENCY2 instead

peterbarker, 2021-07-30:
This was implemented in #17390

closed 2021-07-30 by peterbarker