ArduPilot/ardupilot #7348: SITL build fails with uninitialized frame_proprietary in SIM_Submarine

guglie, 2017-12-04:
#### Issue details
sim_vehicle.py build fails (I'm running it with `-c` option to clean), see log below.

#### Version
current master https://github.com/ArduPilot/ardupilot/commit/87658e6a07545a105e56ed86bec77d9d95b90786

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ x ] Submarine (SITL)

#### Airframe type
None

#### Hardware type
None, SITL

#### Logs
```
../../libraries/SITL/SIM_Submarine.cpp:38:12: fatal error: constructor for 'SITL::Submarine' must explicitly initialize the const member 'frame_proprietary'
Submarine::Submarine(const char *home_str, const char *frame_str) :
           ^
../../libraries/SITL/SIM_Submarine.h:56:7: note: 'frame_proprietary' declared here
    } frame_proprietary;
```

guglie, 2017-12-04:
Maybe @patrickelectric you can help on this? https://github.com/ArduPilot/ardupilot/commit/28ff272017474c77fbfcb16b1c6f35532e160a89

davidbuzz, 2018-03-06:
I just encountered this issue.   Was the original reporter of this using OSX ..?

patrickelectric, 2018-03-06:
@guglie @davidbuzz I can't duplicate it with gcc and clang. What compiler version are you using and which commit hash ?

davidbuzz, 2018-03-06:
I'm compiling with ardupilot 'master' as as of right now  ( 35a4748c06 ) , and it fails on OSX, and I've just  started up a linux VM  and using the same codebase it compiles fine.  So this is clearly an OSX specific problem.  @patrickelectric  - what OS are you using..?

patrickelectric, 2018-03-06:
I'm using clang 5.0.1 and gcc 7.3.0, both compile without problems.

davidbuzz, 2018-03-06:
@patrickelectric  - you still didn't tell us what OS you are using...?

patrickelectric, 2018-03-06:
The code isn't OS dependent, if you run with the same compile and libraries you'll have the same result. I'm running linux.

lvale, 2018-03-06:
I tested this using also master

>  ./waf distclean
>  git submodule deinit -f . 
> git clean -f -x -d 
> git pull master 
> git submodule update --init --recursive 
> ./waf configure --board px4-v4 --check-cxx-compiler g++
> ./waf sub
> 

and no issues

```
BUILD SUMMARY
Build directory: /Users/Luis/px4/ardupilot/build/px4-v4
Target       Text    Data  BSS    Total 
----------------------------------------
bin/ardusub  871440  2772  55864  930076

PX4
The ELF files are pointed by the path in the "Target" column. The .px4 files are in
the same directory of their corresponding ELF files.

You can use the option --upload to upload the firmware to the PX4 board if you
have one connected.

Build commands will be stored in build/px4-v4/compile_commands.json
'sub' finished successfully (3m14.192s)

```

peterbarker, 2018-03-06:
On Tue, 6 Mar 2018, Luis Vale Gonçalves wrote:

> I tested this using also master

Could someone test this (possible) fix on MacOSX, please?

https://github.com/peterbarker/ardupilot/tree/sub-const-struct-fix

patrickelectric, 2018-03-06:
@peterbarker MacOSX have a different clang version from the one that exist in linux repository ?

lvale, 2018-03-06:
with your fix

[image]

davidbuzz, 2018-03-07:
I have tested peter's proposed fix, on OSX, and it's still giving the same basic error as before:
```
[566/971] Compiling libraries/SITL/SIM_Submarine.cpp
../../libraries/SITL/SIM_Submarine.cpp:38:12: fatal error: constructor for 'SITL::Submarine' must explicitly initialize the const member 'frame_proprietary'
Submarine::Submarine(const char *home_str, const char *frame_str) :
           ^
../../libraries/SITL/SIM_Submarine.h:59:7: note: 'frame_proprietary' declared here
    } frame_proprietary;
      ^
1 error generated.
```
II'll also note that my clang version is:
```
 /usr/bin/clang  --version
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
```

lvale, 2018-03-07:
@davidbuzz 

```
Sparkle:bin Luis$ clang --version
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

patrickelectric, 2018-08-25:
@davidbuzz I'm building now with Mac OS, everything is fine here. 
Are you still having this problem ?

davidbuzz, 2018-08-25:
I no longer run osx, so can't say. Unless original poster says, I recommend closing the issue

guglie, 2018-08-27:
@patrickelectric Tested now, no, I'm not having it anymore. Probably I had an old compiler, I changed computer in the mean time. Thanks!

closed 2018-08-27 by guglie