ArduPilot/ardupilot #7350: sim_vehicle.py build failing due to missing termbits.h

guglie, 2017-12-04:
#### Issue details
`asm/termbits.h` is not found during build, I think it's a portability problem since I'm on Mac.

#### Version
Latest master

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
-

#### Hardware type
None, SITL on Mac

#### Logs
```
../../libraries/AP_HAL_SITL/UART_utils.cpp:25:10: fatal error: 'asm/termbits.h' file not found
#include <asm/termbits.h>
```

guglie, 2017-12-04:
If someone knows a replacement library I can do the fix. Maybe @tridge or @magicrub that worked on this? da4d3ebe2bf15dec4ae32a924a2bfbd2d09a44bc

khancyr, 2017-12-04:
It think the simplier for now should be to do like for cygwin and ignore this feature

guglie, 2017-12-04:
Sure, good point

lvale, 2018-03-28:
Following the most recent tutorial every build works fine

http://ardupilot.org/dev/docs/building-setup-mac.html

closed 2018-03-28 by lvale