ArduPilot/ardupilot #7354: Copter: add pre-arm check if frame can't find the output for a motor

rmackay9, 2017-12-05:
In Copter-3.5 (and higher) the AP_Motors library requires that the SERVOx_FUNCTION be set for each motor.  Although these parameter default to good values, it's possible that a user could mess up the SERVOx_FUNCTION settings so that there are no defined outputs for some motors.  We should raise a pre-arm check failure in such cases.

WLeeeee, 2018-02-15:
Hi
I am interested in this one, but I cannot find anything to do with SERVOx_FUNCTION. Could you please tell me where is it? Thanks

WLeeeee, 2018-02-15:
Hi 
I am interested in this issue.
I guess I have add a parameter check in AP_Arming_Copter::parameter_checks, is it?
Thanks

MohamedAliRashad, 2019-03-21:
Is this considered Fixed ?

[image]

shbhmexe, 2025-10-26:
Hi, could you please assign this issue to me? I’ve reviewed the project’s Code of Conduct and Contribution Guidelines, and I have the required skills and understanding of the tech stack to work on this effectively. I’ve already started analyzing the codebase and begun initial work toward resolving the issue. I’m confident I can implement a clean, optimized, and maintainable solution that aligns with the project’s standards. Looking forward to contributing!