ArduPilot/ardupilot #7360: Copter: D-GPS precision landing

DavidIngraham, 2017-12-05:
#### Issue details

The UBLOX M8P supports moving baseline measurements, which gives a highly accurate vector between the rover and base. For certain applications, D-GPS based precision landing could be used for precision landing on a moving platform such as a slow moving delivery vehicle or a boat.

The AP_PrecLand library supports precision landing for copters using a variety of sensor types. An additional backend for this library could be added to support prec-landing using D-GPS.

Another backend could also be added to support DWM-1000 Beacons.

SpektreWorks would be happy to provide a HERE+ kit to a dev willing to take this project on.

#### Version
N/A (Enhancement)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
N/A

#### Hardware type
All

#### Logs
N/A

geofrancis, 2017-12-05:
Let me see if I understand this you want to use another GPS as a target for landing?

DavidIngraham, 2017-12-05:
Essentially, yes.

We would like have a HERE+ Base unit with a companion computer attached to the landing target. This would stream its location and RTK corrections to the vehicle, which would have a HERE+ Rover unit.

This would give the vehicle a very accurate vector to the landing target which would be updated at a high frequency.

rmackay9, 2017-12-05:
extending precision landing to accept and external position via mavlink seems like a great idea to me.
I'm not sure the UBlox RTK unit will work well if the "rover unit" is moving around though.  I think for RTK the base unit needs to be still.

So I think we would need either the ground station or a simple companion computer on the ground to send the home position at a high rate to the copter... and then we modify the precision landing to use that home position for the final phase of the land,

OXINARF, 2017-12-05:
@rmackay9 What @DavidIngraham is talking about is using the units in the moving baseline mode. That gives an accurate vector between the two units.

@DavidIngraham AFAIK using that mode in a M8P will lose the capability to give centimeter absolute position, would that be OK?
@DavidIngraham Regarding the DWM1000 beacons, we kind of support them through the Pozyx solution, but how do you envision it being used? It gives a relative position of the vehicle, but it doesn't say where a target is.

DavidIngraham, 2017-12-06:
The most recent firmware update of the M8P supports a "Moving Baseline" Mode. See page 87 of [this document](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29.pdf)

As @OXINARF suggested, in moving baseline mode you do not maintain absolute accuracy, but have very good relative accuracy between the two units. This would be fine for our use case. When using moving baseline mode, the absolute position estimate should have the same accuracy as a normal M8N.

@OXINARF The DWM1000 note was a bit of an afterthought, but I was thinking that a similar solution to posyx could be used to give the vehicle a relative position in the reference frame of the landing target. 

The POZYX Base tags would have to be connected to fixed points on the moving target, and a landing point would be defined in their coordinate frame. The vehicle or companion computer would then calculate the vector between the vehicle's current position and the desired landing point and then feed that into the PrecLand Library.

DavidIngraham, 2017-12-06:
@WickedShell Do you have any input on the AP_GPS modifications necessary to use the M8P in moving baseline mode?

OXINARF, 2017-12-06:
@DavidIngraham Interesting idea about the DWM1000, probably needs quite a bit of work.

The GPS library should be relatively easy to modify I think. Connecting AP_PrecLand with AP_GPS is probably the hard part (to have a nice solution and not hacking it in).

WickedShell, 2018-01-25:
@DavidIngraham it's a requested and needed feature. However I admit I'm mostly tracking the usage for the GPS heading case first. If you read through the associated documents with the moving baseline mode there are some strict lag requirements to not fall back to extrapolated data, the main one being that the latency between getting the data from one GPS to the other GPS must be < half the update rate of the GPS. For us that means a 100 msec lag term. I'm quite comfortable that this can be done without a stretch on the for the compassing use case where we are directly wired to both GPS's, but I'm not as confident in the offboard version.

It will also require some testing as the moving base GPS is now essentially stuck waiting for correction data, so the EKF will be seeing a more lagged input from that GPS. PPS support could clean the lag up nicely, but that is not currently supported either.

As far as the landing use case goes though I don't think this gets you enough data by itself. We do not want to land on the GPS antenna directly, because we will shadow it, and cause reception problems which would in turn ruin the performance of the moving baseline. However if your GPS antenna is offset from the desired landing point (let us say by a meter) then the aircraft will need to also know the distance and the current bearing from the baseline point that it should be targeting to land since the vehicle is specified as moving, and thus possibly turning/changing heading as well. I'm not sure if we have a means of readily accepting this data in the precision landing library.

rrr6399, 2018-06-02:
FYI: I just added support to pull the RTK solution vector from the M8P to generate the  GPS_RTK record.  I'm looking into how to quickly use this to create a precision follow-me mode that could be used for landing on a moving platform. I'm currently just trying to get the moving baseline mode working on the M8P to see what kind of relative and absolute results to expect. 

I have wondered if I specified the guided waypoint using 
SET_POSITION_TARGET_LOCAL_NED using the MAV_FRAME_LOCAL_NED or MAV_FRAME_BODY_NED coordinate frame using the current base-to-rover measurements coming back from the drone’s RTK GPS in the GPS_RTK message would work. 

Does anybody know if the positions of the guided waypoints can be specified in relative coordinates to the body rather than absolute WGS84 coordinates?