ArduPilot/ardupilot #7374: Rover: LED becomes green before EKF origin has been set

rmackay9, 2017-12-07:
The RGB LED should only turn green once the GPS is good *and* the EKF origin has been set.  So really the LED is telling the user whether we have a good position estimate and thus whether they can use the autonomous  modes.

Of course we risk users complaining that GPS lock takes longer than it use to but it's still more useful because a GPS lock alone is not enough.

WickedShell, 2017-12-07:
I'm not a fan of the implication that the LED indicators are dependent upon platform/use case versus a standard meaning. If this is really important to flag to the user could this be an alternate flash mode (say blue/green alternating?)

rmackay9, 2017-12-07:
@WickedShell, I agree we should be consistent across the vehicles.  For Copter and Rover I think we'd be happy with the LEDs only going green after the EKF's position estimate is good... but if that's not acceptable for Plane then a new pattern sounds fine to me.

kd0aij, 2021-12-21:
What's the status of this issue? I have a use case (plane with no GCS and ARMING_REQUIRE=0) in which it is important to know whether the EKF is ready, and currently that is not possible.

rmackay9, 2021-12-22:
I'm surprised to find that this is still an issue.  I just tested it on my CubeOrange and while the vehicle is in Manual mode the LEDs turn green even before the EKF origin is set.

Since this issue was raised we've improved Rover's arming checks so vehicles can only be armed once the position estimate is ready if the mode requires a position estimate.. so it isn't a safety issue, just a display issue (for Rover).

kd0aij, 2021-12-22:
Same for plane, if you require arming.  if you don't, as with my black-box loggers (and arming is apparently necessary for the EKF to stop moving the NED origin around) then there's no audible or visible indication that the EKF is ready to fly.

It also means that even with arming required, if you don't have a GCS handy, the only signal that you have that the plane is ready to arm, is the ready-to-arm tonealarm.  So it seems to me that we *should* change solid green to mean ready-to-fly instead of 3D fix.

kd0aij, 2021-12-22:
BTW, what is the use case for the "traffic-light" color sequence?
And perhaps this issue is best addressed by adding yet another for my black-box logger use case.

rmackay9, 2021-12-22:
@kd0aij, if I understand correctly, I think the "traffic light" colour sequence is for calibration. So when the gyro calibration is being done soon after startup we see the LEDs going through some sequence of colours (red+blue+green(?))

peterbarker, 2024-07-31:
Well, it's 2024 and we accidentally closed this issue with https://github.com/ArduPilot/ardupilot/pull/27385

Yay :-)

closed 2024-07-31 by peterbarker