ArduPilot/ardupilot #7375: if there is no ToshibaLed  zombie dirver eats memory

night-ghost, 2017-12-07:
#### Issue details
file AP_Notify/ToshibaLED_I2C.cpp line 56. We just created I2C device driver but 1st transfer fails so we choose to "return false".  But stop! driver still alive (no GC!) so just eats memory.

Needed "delete _dev;" before return

#### Version
git

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

OXINARF, 2017-12-07:
Do you mean at https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/AP_Notify.cpp#L234?

night-ghost, 2017-12-07:
ToshibaLED_I2C.cpp is a part of AP_Notify, but I says what happens in ToshibaLED_I2C.cpp itself. Function hw_init() should be started like
```
bool ToshibaLED_I2C::hw_init(void)
{
    // first look for led on external bus
    _dev = std::move(hal.i2c_mgr->get_device(_bus, TOSHIBA_LED_I2C_ADDR));
    if (!_dev || !_dev->get_semaphore()->take(HAL_SEMAPHORE_BLOCK_FOREVER)) {
        return false;
    }

    _dev->set_retries(10);

    // enable the led
    bool ret = _dev->write_register(TOSHIBA_LED_ENABLE, 0x03);
    if (!ret) {
        _dev->get_semaphore()->give();
        delete _dev; // @NG - don't let zombie to eat memory
        return false;
    }
```

night-ghost, 2017-12-08:
Sorry, _dev is OwnPtr so should be _dev=nullptr instead delete

night-ghost, 2017-12-08:
or full AP_Notify should be rewritten to call static methods probe() from individual drivers, as it now does compass and baro drivers

magicrub, 2017-12-08:
pull-requests welcome!

night-ghost, 2017-12-08:
sorry but no, I already tried :)

It is easier for me to maintain my own fork with fixed bugs, than to do lots of endless corrections, and after all that torments PRs hangs a whole year without moving. Yes I understand - the Policy :)

PS. if you do not like that I'm trying to help you fix bugs - I will stop these efforts, just let me know.

CraigElder, 2017-12-08:
Don't stop trying with PRs. By definition what we do here is make endless corrections. 
We know we have a backlog. We could use some help.

Pedals2Paddles, 2017-12-08:
If your PR needs corrections, I'm not sure how that is the fault of ArduPilot's dev team, or an example of you being held back.  The standards are strict, as they should be. Francisco tortures me nitpicking every PR I do for every little thing.  As he should!  If my code has errors, no matter how small, I need to fix it before it goes in.  It annoys the hell out of me. But it's necessary.  And the result is the code I submit is clean, correct, and professional.

I did a search, and I see only a few PRs submitted by you.  None were ignored or left hanging. They got lots of attention from the dev team, providing input on the corrections and changes you needed to make.  If you do not want to make those corrections... or you do not want to rebase the changes over time, that's squarely on you.  It's the nature of developing.

night-ghost, 2017-12-12:
>Don't stop trying with PRs.

Life is one, and I'd rather will do something more useful than useless efforts in PR here.

>If your PR needs corrections

But if all required corrections are done, but PR hangs for year?

>If my code has errors, no matter how small, I need to fix it before it goes in. 

Sure, but what if there are no errors? Code works, people flying and happy - but no, we need to change almost all just to add new bugs... Personal style preferences, architecture preferences - yes, but not errors.

Pedals2Paddles, 2017-12-12:
These are not personal style preferences or simply making busy work.  Believe me, everyone here has way better things to do than fabricate busy work. These are the standards by which all code in ArduPilot is held to.  Your PR had a LOT of work still needed.  And the dev team was very actively working through them with you.  The PR didn't hang for a year. You gave up on it and walked away before it was done because you didn't want to adhere to the standards that everyone else is held to.

I can write a bunch of sloppy code that "works".  That doesn't means it is acceptable and should be merged.  Yes it is time consuming but that time pays off because it's done right. If you simply do not have the time to do it, that's unfortunate.  If you have the time but simply don't like it, that's even more unfortunate.

auturgy, 2017-12-12:
NG,
thanks for pointing this out.
re PR’s, I can understand your frustration. Matt makes some valid points, but I also think we need to make it easier/clearer for contributors, to minimise this type of thing and make ArduPilot as welcoming as possible. I for one appreciate your considerable efforts, and hope you keep at it.

night-ghost, 2017-12-12:
>These are not personal style preferences or simply making busy work.

sorry, but it's just an insulting noise. I have my own opinion, you have your own, and no one will agree to take a step forward.

Pedals2Paddles, 2017-12-12:
Coding standards and quality control are not "insulting noise". Nor is it an opinion.  If that's how you feel... that you should be able to submit whatever you want to ArduPilot and get it merged as long as you managed to make it compile... than it may be for the best that you don't.

hiro2233, 2017-12-24:
Hello, this https://github.com/ArduPilot/ardupilot/pull/7453 solve the problem.

@night-ghost  Don't worry be happy, i'm trusting on the apm dev team, and i'm learning too much with the development, maybe there are people on the dev that is no pretty, for example like Francisco :v but i think that we are doing the best for the future of the development. In my country i'm growing up thanks to APM and i'm grateful for that.

night-ghost, 2017-12-24:
>Don't worry be happy

so please eat shit very snappy, right? :) Thanks I promise to not correct  bugs any more

khancyr, 2017-12-24:
Please stop insult, we are here to speak about developpement and coding for drones, not for anything else.

hiro2233, 2017-12-25:
@khancyr really? so what i said that you consider an insult? you are trying to make it personal?

Pedals2Paddles, 2017-12-25:
@hiro2233, he was talking to night ghost, not you.

khancyr, 2017-12-25:
@Pedals2Paddles, it was adressed to Hiro.
 But case closed. I will remove unrelated message to this topic next to this message.

hiro2233, 2017-12-25:
💩

Pedals2Paddles, 2017-12-26:
Clean up the childish insults and name calling.  So let's keep this on-topic.