ArduPilot/ardupilot #7379: ArduPlane/quadplane: TECS causing alarming altitude loss

expntly, 2017-12-09:
#### Issue details
During an AUTO mission, an in-flight issue caused our horizontal propulsion to produce 0 thrust.  Supposedly because of TECS, altitude was traded for speed.  So the aircraft went from 100m AGL to 38m AGL, when we had to take over manually in QSTABILIZE/QLOITER.

1) what would have happened if we hadn't taken over manually?

2) in the same way the Q_ASSIST_* parameters work, is there a way the FC could recover in VTOL when such a situation occurs and when it is too close to the ground?  Maybe we're looking for something like a, say, Q_ASSIST_MIN_ALTITUDE?

TECS_ params for reference:

> TECS_APPR_SMAX,0
> TECS_CLMB_MAX,5
> TECS_HGT_OMEGA,3
> TECS_INTEG_GAIN,0.1
> TECS_LAND_ARSPD,-1
> TECS_LAND_DAMP,0.5
> TECS_LAND_IGAIN,0
> TECS_LAND_PDAMP,0
> TECS_LAND_PMAX,10
> TECS_LAND_SINK,0.25
> TECS_LAND_SPDWGT,-1
> TECS_LAND_SRC,0
> TECS_LAND_TCONST,2
> TECS_LAND_TDAMP,0
> TECS_LAND_THR,-1
> TECS_PITCH_MAX,0
> TECS_PITCH_MIN,0
> TECS_PTCH_DAMP,0
> TECS_RLL2THR,10
> TECS_SINK_MAX,5
> TECS_SINK_MIN,2
> TECS_SPD_OMEGA,2
> TECS_SPDWEIGHT,1
> TECS_SYNAIRSPEED,0
> TECS_THR_DAMP,0.5
> TECS_TIME_CONST,5
> TECS_TKOFF_IGAIN,0
> TECS_VERT_ACC,7

#### Version
3.8.2-dev

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
QuadPlane X8

#### Hardware type
PH2.1

#### Logs

magicrub, 2017-12-09:
Q assist does this, without a log I will assume you disabled it. Also, this is a help questionswas, not a confirmed big. This is better posted on the forums with a log. Closing.

expntly, 2017-12-11:
It is more of a feature request issue so as to keep track.  But if Q assist does this then all you have to do is let us know how before closing, thanks :-)

magicrub, 2017-12-11:
Q_ASSIST_SPEED will turn on the vtol motors if you're too slow, Q_ASSIST_ANGLE will turn on vtol motors if your actual attitude diverges from the desired which is common during a stall. Basically, the Q_ASSIST_* params make your aircraft stall-proof
http://ardupilot.org/plane/docs/quadplane-flying.html#assisted-fixed-wing-flight

expntly, 2017-12-13:
Neither of these happened during our flight, so Q_ASSIST didn't kick in.  Maybe I miscommunicated or misunderstood what you said initially.

We're basically looking for a way to prevent crashing into the ground because of TECS -- so maybe it's a bug or maybe it's a feature request.  In a quadplane that's flying cruise, that could mean de-transitioning so that the plane can get back to VTOL.

tridge, 2017-12-13:
yes, we should fix this with a Q_ASSIST_ALT parameter I think. We'd then need to work out what it should do when it triggers though. Should it land straight down?

expntly, 2017-12-14:
That, or maybe qloiter for a while at that altitude to give a chance to ground to take over (error should be reported on GCS), and then land straight down after a timeout.

Best if this altitude is AGL.

expntly, 2018-01-06:
Until this is supported, is there a way geo-fencing can help?  I'm thinking a low-bound floor, say 50m AGL, when the plane goes lower than that, geo-fencing kicks in and the plane detransitions to VTOL.

IamPete1, 2021-08-15:
closed by https://github.com/ArduPilot/ardupilot/pull/12736

closed 2021-08-15 by IamPete1