ArduPilot/ardupilot #7385: Rover: use wheel encoders in motor control loops

rmackay9, 2017-12-10:
Rover 3.2 supports wheel encoders but they are only used for position estimation.  They are not used to improve the vehicle's turn-rate and speed controllers.

The AP_Winch library (used in Copter-3.6+ to lower packages) includes some of what we need.  In particular it uses the wheel encoder output and adds a PID controller to better control the speed at which the winch line is released or retracted.

mikerob, 2017-12-11:
Is this an opportunity to also allow use of single-line encoders for wheel speed (rather than only quadrature)? 
I know you'd really need two quadrature encoders on a skid steer vehicle, but for conventional steering you only need one wheel signal with direction, and the others could be single line. That would let you encode all four wheels with the available inputs.
That, in turn, eventually opens up traction control (comparing driven to non-driven wheel speed) and anti-skid braking.

rmackay9, 2019-03-13:
This has been included in Rover-3.5.  The WRC_ parameters can be used to set this up.  We should document this though.

closed 2019-03-13 by rmackay9