ArduPilot/ardupilot #7386: Rover: combine SPEED_TURN_GAIN and WP_OVERSHOOT

rmackay9, 2017-12-11:
Rover has two separate methods for reducing speed to improve cornering in Auto, Guide and RTL.  We should consolidate these methods somehow.

The WP_OVERSHOOT parameter is set by the user and the vehicle should slowdown as it reaches a waypoint so that the vehicle will not overshoot the waypoint by more than this distance (in meters).  The calculations for the _desired_final_speed value can be found in mode.cpp's Mode::set_desired_location method (https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.cpp#L128)

The SPEED_TURN_GAIN parameter allows the user to set a minimum speed (expressed as a percentage of the desired speed (which normally comes from CRUISE_SPEED, WP_SPEED or RTL_SPEED) used during turns (in Auto, Guided, RTL).  The final speed used during cornering uses linear interpolation between this minimum speed (i.e. SPEED_TURN_GAIN) and the desired speed (i.e. CRUISE_SPEED or WP_SPEED) based on the desired lateral acceleration (from the L1 controller) as a percentage of the maximum lateral acceleration (i.e. TURN_MAX_G) and the heading error between the vehicle's current heading and the heading to the next waypoint.
Code can be found in mode.cpp's Mode::calc_reduced_speed_for_turn_or_distance.  https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.cpp#L128

I have two rough ideas:

- create a combined wp.update function that calculates both desired-throttle and desired-lateral-acceleration for this iteration.  It might pass in the desired-speed to the L1 controller instead of using the current vehicle speed.  It might immediately scale back the speed so the vehicle's lateral acceleration stays below the TURN_MAX_G parameter value.  Maybe it would handle low-speed maneuvers differently - cases where the lateral acceleration controller may be having trouble.
- it is probably easier to remove WP_OVERSHOOT and instead, set the _desired_final_speed using an estimate of what the calc-reduced-speed-for-turn-or-distance will do when we reach the waypoint.  We could assume that when the vehicle reaches the next waypoint it will be pointed along the line from the current origin to the destination.  We could use the L1 controller's P gain (or whatever it's called) to calculate what lateral acceleration it will request as we pass the destination and head towards the next waypoint.

rmackay9, 2017-12-11:
After a ping from @peterbarker, I should add that there is another parameter involved for cases involving pivot turn vehicles, PIVOT_TURN_ANGLE.  So if the heading error (i.e. difference between the vehicle's current heading and the heading to the next waypoint) is larger than this parameter value the vehicle reduces it's speed to zero.  This logic is also in the Mode::calc_reduced_speed_for_turn_or_distance function.

khancyr, 2017-12-11:
Perhaps we could include better motion and trajectory control with better acc and deceleration phase calculated from the next waypoint.

Oussalvatore, 2018-04-05:
Please, what is the equivalent of this configuration in the case of  Copter ??