ArduPilot/ardupilot #7393: ArduPlane/quadplane: vehicle jerks when executing VTOL_LAND

expntly, 2017-12-11:
#### Issue details
In an AUTO mission, as soon as the vehicle reaches the VTOL_LAND waypoint, it looks like it switches controller and jerks for a couple of seconds, causing high pitch+roll+yaw, until it gets back to level and starts descending.

[image]

#### Version
3.8.2-dev

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X8 quadplane

#### Hardware type
PH2.1

DavidIngraham, 2018-02-07:
Can you provide a log of the flight? Also, I'm not sure that I understand exactly what the problem is.

IamPete1, 2021-08-15:
not much we can do without the log, if this is still a issue please re-open and share a 4.1 log.

closed 2021-08-15 by IamPete1