ArduPilot/ardupilot #7396: HIL_GPS message handling

JustFineD, 2017-12-12:
The HIL_GPS message (#113) defines the altitude field as [m]*1000

In the implementation of AP_GPS_MAV::handle_msg(...) - the altitude is copied without a conversion into gps state :
...
loc.alt = packet.alt;
...

Isn't the loc.alt should be given in [cm] ?
loc.alt <- packet.alt / 10.0f  // [mm] -> [cm]

WickedShell, 2018-03-28:
Tagged as dev call topic, since we've been fixing other issue similairly. Does anyone know if MAVros works around this already or is it causing people issues? @khancyr 

(offtopic: we need to work to push people towards the GPS_INPUT message over HIL_GPS since it supports finer grained controls/fixes a few other things.

vincekurtz, 2018-04-02:
Not sure if it's the same issue, but I've been having [some difficulty](https://discuss.ardupilot.org/t/flight-mode-position-hold-refused-despite-strong-gps-signal/27046) using MAVros and the HIL_GPS message: namely not being able to enter position-hold mode despite accurate positioning from a mocap system. 

Also, the ground speed calculation seems a bit funny to me:

```
if (packet.vel < 65535) {
    state.ground_speed = packet.vel  / 100.0f;
}
```
Why not just use `state.ground_speed = norm(vel.x, vel.y);` like for the GPS_INPUT case?

WickedShell, 2018-04-03:
@vincekurtz There isn't a need to, as the MAVLink spec states that packet.vel is actually specified as the ground speed velocity, rather then the 3D velocity.

WickedShell, 2018-04-03:
Corrected in master.

closed 2018-04-03 by WickedShell