ArduPilot/ardupilot #7399: SAFETY CRITICAL: add a check if the external mag is OK.

sashan13, 2017-12-12:
#### Issue details

So the issue is as following. I almost had a crash today. The reason turned out to be a mechanical damage to external magnetometer mounted at that small GY board. Pixhawk reverted to internal mag but it was a silent process that happened without any sorts of messages or errors. Due to copter being stationary on the ground all the way to takeoff it never detected the problem using EKF variance. Upon takeoff I had a typical compass interference problem with copter going crazy, toiled bowl effect et cetera.

What we really need is a way to detect such problems. There should be a way to disable compass autodetect after initial setup, alert user that something is amiss should configuration become different from saved one, and fail pre-arm check. 

#### Version
3.5.4 or whatever the previous one from current is.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad with carbon fiber shell fuselage.

#### Hardware type
Pixhack

#### Logs
Not sure if needed.

rmackay9, 2017-12-13:
If the arming checks are on, it should catch this problem.  The way it does this is during compass calibration it records all the device ids of the compasses (these ids are not unique to the individual compass but are unique for the type of compass and port used).  Then as part of the arming checks it compares the currently detected compasses (including their order) with the list recorded during the compass calibration.  If the external compass had become detached or inoperable, the lists would not match and a "compass calibration required" arming messages should have been displayed.

A log would be good because it would show the device ids, the compass health and whether the arming checks were on or not.

rmackay9, 2017-12-23:
I haven't heard back on this so closing.  I'm pretty sure we have the checks in place to capture this issue as described above.

closed 2017-12-23 by rmackay9