ArduPilot/ardupilot #7420: Rover: support optical flow

rmackay9, 2017-12-15:
At least one user has asked that we add support for optical flow in Rover.  With some upcoming changes from Paul, we should be able to support optical flow sensors facing in any direction which will likely make it more useful for Rover.

MDB22, 2019-04-04:
@rmackay9 any word on whether Rover does/will support PX4 Flow?

rmackay9, 2019-04-04:
@MDB22, It hasn't been added yet.  I'm not actively working on it actually, for me there are a number of higher priority tasks including improving object avoidance so I can't make any promises on when this will be added.  Maybe another developer will take this task on .. I can't say really..

MDB22, 2019-04-04:
@rmackay9 thanks for your response, I saw a separate thread with several developers making progress on it, but nothing concrete for the main branch. Thanks anyway :)

rmackay9, 2022-11-30:
After only 5 years this task has finally be completed with PR https://github.com/ArduPilot/ardupilot/pull/21510.  This will go out with Rover-4.4.0 which should enter beta testing in early Jan 2023.

closed 2022-11-30 by rmackay9