ArduPilot/ardupilot #7427: failsave RTL extended option?

B737-300, 2017-12-17:
// I don't know if this ist the right spot for this- if not, please correct me!!!

I want to have a new RTL feature for AIRCRAFT that does following:

**1st: If failsave is triggered then start climb 50 meters higher straight ahead.
2nd: THEN start the turn to home position or go to the landing sequence.**

2 reasons: at low altitudes there could be trees on the left or right side and so this could avoid that the plane makes the turn direct into.
The other reason is if you fly behind a hill and you loose the TX und Video link you will get it back sooner.

Another feature: If failsave triggers RTL set RTLALT to current alt+50 Meters to make sure that the plane doesn't hit a hill at coming home. 
A distance depending RTLALT could be a fine thing too: 
if DIST >1km RTLALT = CURRENTALT + 50
if DIST < 1km RTLALT = 150m
if DIST < 500m RTLALT =100m
or similar to this.

I don't understand why we still have to use a SBUS converter. As far as I know the problem is that some RX sends SBUS and some inverted SBUS. Isn't it possible to dedect the LOW or HIGH signal?
FYI: I'm using several planes with MINIAPM which isn't supported any longer. (TARANIS+, D4RII, ARDUINO NANO as a converter for MAVLINK to FRSKY and some LUA Scripts)
I would like to change to PIXHAWK if there is a LIGHTWEIGHT and inexpencive hardware avialiable that doesn't need a converter for the SBUS. (Each extra hardware could lead to problems)

And most important: THANK YOU FOR THIS AMAZING SOFTWARE!!!!!!!!
best regards, Richard

night-ghost, 2017-12-17:
>I would like to change to PIXHAWK if there is a LIGHTWEIGHT and inexpencive hardware avialiable that doesn't need a converter for the SBUS.

see here - https://www.rcgroups.com/forums/showpost.php?p=38820459&postcount=1801

You can use ArduPlane on almost any F4 FC, with autodetection of SBUS polarity

B737-300, 2017-12-17:
Thanks for information, but something semms wrong with this link. ;(

night-ghost, 2017-12-17:
another link - https://www.rcgroups.com/forums/showpost.php?p=38825435&postcount=1804

B737-300, 2017-12-17:
To: "night-ghost" 
Was soll in dem link für eine Info sein? Die 2 Fotos sind cool, aber irgendwas versteh ich nicht. Hab mit einer Liste von FC's gerechnet, auf denen ARDUPLANE mit SBUS autodedection funzt-oder so. Sorry, falls ich da was nicht kapiere! Ich weiß, daß man auf dem APM board den Chip (nicht den Main Prozessor) umflashen kann, um das invertierte SBUS Signal zu verarbeiten. Die ganzen F4 controller surren ja kiloweise auf jeder Wiese herum. Werden die von MISSION PLANNER auch als kompatibel erkannt? Meine Idee ist es auf  PIXHAWK umzusteigen, um 2 Telemetrie Ports zu haben- MINIMOSD und FRSKY Telemetrie. Das zusätzliche ARDUINO Board ist mit den verschiedenen OPENTX Versionen zu mühsam geworden und ja irgendwie "unnötig". 
Zumindest sind die F4 controller viiiiiiel kleiner und günstiger und in X Versionen erhältlich- wenn das auch ginge wäre es fein.
Wollte eigentlich nur das mit dem link auf Deutsch klären!- ist jetzt ausgeartet ;)

geofrancis, 2017-12-17:
all pixhawk controllers can take inverted sbus without any adapters.

B737-300, 2017-12-17:
That means that I can use it directly with FrSky Recievrs that have SBUS?
What about theese many posts about an adapter with a diode?? I'm confused.
Please tell me more!

geofrancis, 2017-12-18:
there is nothing to tell, you just plug the receiver into the pixhawk with sbus and it will work, you might be talking about the telemetry adapter, that is a a TTL-RS232 adapter with a diode.

geofrancis, 2017-12-18:
This should be closed

auturgy, 2017-12-18:
Geofrancis is correct: sbus works out of the box. Frsky telemetry via a uart requires an inverter for most setups.
Re the original request: does smartRTL provide enough of what you need?
I’d like to close this issue as it strayed off topic, but need to be sure your feature request is captured or satisfied

B737-300, 2017-12-18:
Yes- It's far off topic- don't know how this happened.
The main thing of my request was not captured at all- so I don't know what
do to best.

B737-300, 2017-12-18:
Sorry for the confusion- I was to far in "telemetry problems" theese days.
So thanks to everybody for confirming that "the inverted SBUS" is no problem for PIXHAWK.

This could be a inexpencive F4 board: 
(But I didn't find any reviews until now- semms to be very new)
https://www.banggood.com/Radiolink-Mini-PIX-F4-Flight-Controller-MPU6500-w-TS100-M8N-GPS-UBX-M8030-For-RC-Drone-FPV-Racing-p-1240423.html?rmmds=search&cur_warehouse=CN

The whole thread became off topic- I think there were to many questions in my original post- sorry for that! AND: Thanks for your patience!!!
I don't know what do do now- the main request was to improve the RTL behavior.

auturgy, 2017-12-18:
Perhaps open a new issue, with just a simple description of the feature request and I’ll then close this one to keep things clear.

Regards,

James

B737-300, 2017-12-19:
That souds good to me.
Thanks!

closed 2017-12-20 by lvale