ArduPilot/ardupilot #7433: Rover: move Reset Switch to auxiliary function

rmackay9, 2017-12-20:
There is a "Reset Switch" feature in Rover that sets the flight mode based on the position of the transmitter's mode switch (http://ardupilot.org/rover/docs/common-rc-transmitter-flight-mode-configuration.html).

This is perhaps useful when recovering from a failsafe event in which the failsafe itself has triggered a mode change.  Suppose the rover was in Auto and the failsafe triggered and switched the vehicle into Hold mode.  If the user wanted to switch back into Auto they could either move the transmitter's mode switch to another position and then back to Auto or they could use this "Reset Switch" feature.

I'm not sure I would personally ever use this feature, but in any case, instead of having a separate RST_SWITCH_CH parameter we should just add it as an option to the auxiliary functions channel.  In case anyone complains that they want to make use of two channels (i.e. one for an auxiliary function, another for this "reset switch" feature) we should tackle this by extending the number of channels that can be used for auxiliary functions.

hdtechk, 2017-12-26:
Would really like to see more auxiliary channels.
And someday channel 6 tuning functions like copter

rmackay9, 2019-03-13:
This has gone out with Rover-3.5 (along with more auxiliary channels) so closing

closed 2019-03-13 by rmackay9