ArduPilot/ardupilot #7434: TradHeli: Configurable PWM Pulse width and RC_SPEED *per each servo* (digital/tail servo compatibility)

GlidLov, 2017-12-20:
#### Issue details
Bigger Helicopter uses CCPM (120°) swashplates and can use special tail servos with different pulse witdth than standard (centered at 760us rather than 1500us).
Since RC_SPEED is a global parameter right now, you cannot use this mixed servo configurations. Either, manually setting RC_SPEED can lead to burn expensive servos. 

Pls make per-servo configuration possible, letting mixed freq rates config possible (maybe useful also to other configurations, like quad-plane and so on) and also pls make some preset for "standard size, high performance" helicopter digital servos (333hz, 760us, etc) 

#### Version
3.5 trad heli

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
700-class tradheli

#### Hardware type
Pixhawk/Pixhawk2

lvale, 2018-10-05:
@tridge

ChristopherOlson, 2019-02-26:
@tridge is this still, in your opinion, a valid feature request? We can already set the pulse width individually. I have flown plenty of helicopters with 760us centered tail servos at 125, 200, 333 frame rate, they work fine and it doesn't make one bit of difference what the frame rate is because the rate control loops don't run that fast.

Would like to clean up some of these old issues if we're not going to do anything with it. I don't see any point to this as increasing the frame rate to a servo only updates it more often with static information. I have not found an instance yet where running a 760us x 560Hz servo at 125, 200 or 333 will burn it out.

ChristopherOlson, 2019-02-26:
@bnsgeyer would be good to get your thoughts on this

bnsgeyer, 2019-02-26:
@ChristopherOlson   What do you mean that the control loops don’t run that fast? The control loops run at 400 Hz. Also I would like to see how you’re able to use servos with a 760us center point. I never understood how you can do that setup in ardupilot. 
As for this issue, it appears to me the only thing we are unable to do is set servo speed for each servo.  Don’t know if I have the experience to say that would be useful other than for a tail servo.

ChristopherOlson, 2019-02-26:
The tail servos in question have an update rate up to 560Hz. But these were a "fad" because they only last 10-15 hours. The controller and motor run so hot in them they self-destruct. People bought into the "fad" of faster, quicker, hotter. But to make these servos they had to reduce the pulse width, 760 microsecond center.

So after the "fad" died out now many manufacturers like B-K Servo don't build a 560Hz servo anymore. All their 760us center tail servos are 333's like the cyclics.

It is no problem running 760us center servos in ArduPilot. Simply set the SERVOx_TRIM to 760 and set the SERVOx_MIN/MAX to whatever you need for tail travel on each side of that. They have an operating pwm range of 460 to 1060. If you hook a 760 servo up to a port configured for 1520us center it will burn it out instantly. But otherwise the 760 servos "just work" in ArduPilot.

lvale, 2019-02-27:
I've used the 760us servos on default servo frame rates given that before plugging them we have to go to the servo4 port and set the trim to 760us manually (no gcs likes this) and Pam min and max to whatever values the servo uses (I previously measured max and min pwm the servo uses).

Regarding the frame rate, I believe the problem was that the pins have a frequency by group not individually.

Running at the default frequency the servo I've been using on the tail (have to check the model) has more than 20 hours and doesn't even get warm to the touch.

ChristopherOlson, 2019-02-27:
The 760 servos are fine running them at reasonable frame rates. But bumping them to 560 Hz really shortens their life and they get very hot, even the coreless models.

When it comes to setting frame rates for heli servos, honestly the the only difference I can detect setting frame rate to the default of 125Hz vs 333Hz, is that the cyclic servos will run hotter at 333. But as far as how the helicopter flies - can't tell one bit of difference. And neither can anybody else. This has been discussed ad infinitum on dedicated heli sites like HeliFreak and it has yet to be proven that running servos at higher frequency really has any advantage that a pilot can tell in speed. Mainly because even very fast systems like the 7 ms Futaba transmitters are still only capable of sending 1/0.007 = 143 frames/sec. But creative marketing works to get people to buy into it :-)